SmingHub/Sming #536: Buffering on SPIFFS

andriyadi, 2016-01-06:
I'm doing audio streaming and buffering on RAM is kinda suffering due to limited space. I try to do buffering on SPIFFS by writing to file. Then after certain amount of data are saved to that file, I fire a timer to start reading the data from it and do further audio processing.

I initially wrote and read a big file for buffer, and apparently reading and writing the same file causes not good performance. Then I've tried to use few smaller files, after few files is done writing, I start reading. But it seems writing and reading at the same time even for different files has impact on performance.

Anybody can explain why is that happening or maybe have another thought to do buffering on file system?

Thanks!

tprochazka, 2016-01-09:
Maybe this is good start point
https://github.com/pellepl/spiffs/blob/master/docs/TECH_SPEC
And there is also FAQ: https://github.com/pellepl/spiffs/wiki/FAQ
which can help you.

vsirvent, 2016-01-10:
Hi @andriyadi,

I don't think buffering to flash is a valid solution. Flash is slow and has a limited number of sector writes, so even if you make it work you will break the flash memory after some time. Flash is not designed for continuous read/write in a microcontroller.

You should use RAM memory for audio buffering. I have not measured the jitter when receiving UDP packets in this particular micro, but if the sender is located in the same network it should be low enough to be compensated by buffering 2 or 3  UDP packets, which can be done with the current available RAM. If you need more data to be buffered, may be you should check the sender and try to optimize it (real time tasks? task priority ok? socket options ok? periodic send ok?), check the jitter in the network with wireshark and try to optimize the network (can the UDP data be tagged with QoS?), etc... 

Hope this helps,

Vicente

andriyadi, 2016-01-10:
Hi guys,
Thanks for your responds so far.

I can make it work. What I did is basically storing data to file in flash and making it circular after certain file size (by rewinding the file pointer). But after a while, it starts to call "api_spiffs_erase" function (I knew from debug message) that make writing and reading whole lot slower.

So, I guess it's indeed not for continuous read/write. From the FAQ link provided by tprochazka, writing cycle is limited.

I tried to do buffering to file in SD card, but it's not fast enough. And I often experience disk error when writing.

Buffering on RAM is what's left. Before considering any of above options, first I did buffering on RAM. As audio source is from internet, at 192 kbps audio rate, available RAM cannot keep up to store needed data to play audio smoothly. I did some kind of circular buffer on RAM, but at 192kbps apparently at some point reading is faster than writing. But my code may be not good enough, I don't know :)

Maybe someone can point me to any references to do "right" buffering on RAM, that would be much appreciated. 

Thanks.

vsirvent, 2016-01-10:
Hi, 

api_spiffs_erase might mean that you already broke an internal flash sector. SD card is also flash memory, so you have the same limitation.

The problem is that the sender is from internet and that normally means also that you're using TCP connection, which requires more buffering, in this case you normally need some seconds of audio buffering to avoid audio cuts, that will not fit in internal RAM.

You don't have enough pins to use a typical external sram, but you could use an i2c sram memory as external memory. Anyway the topic has nothing to do with Sming.

In my opinion, you're trying to get too much from this micro. May be you should consider other SoC with more RAM memory available.

Regards,

Vicente

hreintke, 2016-01-10:
@andriyadi : @vsirvent : 
The api_spiffs_erase is expected behavior.
Spiffs implements a flash wearing mechanism. That results in files not always in the same physical location. When a block is "freed" spiffs erases for future use. When using spiffs with updating files that will happen regularly.

vsirvent, 2016-01-10:
Right, anyway flash sectors will end up failing if using flash as memory buffer.

hreintke, 2016-01-10:
Not a Sming functionality issue.

closed 2016-01-10 by hreintke

andriyadi, 2016-01-12:
Yep, not Sming issue. 
I'm aware that I'm abusing this SoC, for the sake of curiosity. Now, I'm back pursuing RAM buffering solution. [Sming RTOS](https://github.com/hreintke/Sming_RTOS_POC) helps me a lot. Can't wait to get that repo gets merged to this repo.

Awesome works guys!