SmingHub/Sming #540: Cannot connect to esp websocket from android native code

alon24, 2016-01-07:
Run the sample HttpServer_WebSocket on esp
Use android native code to try to connect to it, it does not connect (it does connect to test server on the web)

```
gradle:
compile 'org.java-websocket:Java-WebSocket:1.3.0'

Android code 
private void connectWebSocket() { URI uri; try { uri = new URI("ws://10.100.102.152:80/index.html?command=true"); //            uri = new URI("ws://echo.websocket.org"); } catch (URISyntaxException e) { e.printStackTrace(); return; } mWebSocketClient = new WebSocketClient(uri) { @Override public void onOpen(ServerHandshake serverHandshake) { Log.i("Websocket", "Opened"); mWebSocketClient.send("Hello from " + Build.MANUFACTURER + " " + Build.MODEL); } @Override public void onMessage(String s) { //                final String message = s; //                runOnUiThread(new Runnable() { //                    @Override //                    public void run() { //                        TextView textView = (TextView)findViewById(R.id.messages); //                        textView.setText(textView.getText() + "\n" + message); //                    } //                }); } @Override public void onClose(int i, String s, boolean b) { Log.i("Websocket", "Closed " + s); } @Override public void onError(Exception e) { isConnected = false; updateScreenState(); Log.i("Websocket", "Error " + e.getMessage()); } }; mWebSocketClient.connect(); }
```

```
onAccept state: 0 K=1 Free heap size=32936, K=1 +TCP connection timeout updating: 70 -> 90 TcpServer onClient: 10.100.102.102
Item: Name = command, Size = 4, Value = true path=/index.html Host === 10.100.102.152 Sec-WebSocket-Key === j55P1mbQBtOcZ/s8fCaGlA== Sec-WebSocket-Version === 8 Upgrade === websocket parsed Request: GET, nodata TCP received: 175 bytes onReadyToSendData: 1 Default server handler for: '/index.html' found index.html send file: index.html (2768 bytes) response sendHeader Switched to WebSocket Protocol timeout updating: 90 -> 65535 TCP sent: 127 onReadyToSendData: 2

robotiko, 2016-01-08:
Looks the same error as https://github.com/TooTallNate/Java-WebSocket/issues/266
Checked the Sming code but it is not stated what draft version is implemented. 

Should debug as in here to get some hints:
https://github.com/TooTallNate/Java-WebSocket/issues/87

But looks like Android lib issue with Sming version implemented.

alon24, 2016-01-08:
I am closing this becuase I found a lib that works ok with sming, so not an esp sming issue: http://autobahn.ws/android/getstarted/

closed 2016-01-08 by alon24