SmingHub/Sming #547: High latency when publishing MQTT messages

Informatic, 2016-01-12:
I noticed major latency (300-500ms) when publishing messages from Sming with long intervals.
Testing code: http://git.io/vzv5P

Turns out it's probably caused by `send()` called in `MqttClient::staticSendPacket` storing data in a stream and flushing it only when it receives readyToSend event. Simply adding `client->pushAsyncPart();` after sending as mentioned above reduces latency to around 10ms.

In theory `send()` could check if buffer was empty at the time of last readyToSend event and call `pushAsyncPart()` immediately in such case. I tried it with simply adding this in `send()`:

```c++
// [...]
stream->write((const uint8_t*)data, len);

if(asyncTotalLen == asyncTotalSent)
    pushAsyncPart();

asyncTotalLen += len;
// [...]
```

That seems to work, however I haven't tested it much yet, and am not sure if it's the proper way to do this, since I just dove into Sming codebase. I need some comment on that before submitting a proper PR.

slaff, 2016-10-16:
@Informatic Is that still the case? As far as I can say the issue was happening because the MQTT client was initialized BEFORE LWIP was initialized and that caused very odd behaviour.

Informatic, 2016-10-16:
That might be the case, can you share some sample code? Pretty much the same code as in https://github.com/SmingHub/Sming/blob/develop/samples/MqttClient_Hello/app/application.cpp still seems to have this issue for me, as far as i know.

slaff, 2016-10-16:
@Informatic Can you try the following changes:
1. Replace that line in the original example: https://github.com/SmingHub/Sming/blob/develop/samples/MqttClient_Hello/app/application.cpp#L30 with that line: https://github.com/slaff/Sming/blob/tasty/samples/MqttClient_Hello/app/application.cpp#L34.

2. Then in your `init` function add that call:

`mqtt = new MqttClient(MQTT_HOST, MQTT_PORT, onMessageReceived);`

3. Replace all `mqtt.{something}` with `mqtt->{something}`

4. Recompile, test and report.
All these changes make sure that the mqtt object is initialized after the LWIP stuff is initialized.

slaff, 2016-11-24:
@Informatic With the latest version in git that should have been fixed. Can you check and tell us if it is working for you? You can test it with the Basic_HardwarePWM example. See free to reopen this ticket if the issues are still there.

closed 2016-11-24 by slaff

Informatic, 2016-12-04:
Ok. Sitting down now to test it... I'll update you later today.

Informatic, 2016-12-04:
Ok, everything seems to be fine now and latency stays around 8-10ms when publishing to broker also connected via wifi. Thank you very much.

Informatic, 2016-12-04:
Ok. Nevermind. I'm an idiot. Forgot about changing SMING_HOME and tested it again with my own patches.

Test code I posted in the issue still has random 200ms ~ 900ms latency on develop branch.

verybadsoldier, 2017-04-18:
I am also experiencing a high latency when sending MQTT messages to the broker. I am on Sming 3.1.2. Latency is wobbling between very low (~ 50 ms) and quite high (~ 500 ms).
This issue here is closed but I could not find a solution. Is there anything that can be done to fix this? Thank you guys.

verybadsoldier, 2017-12-15:
I am still suffering high latency on 3.5.0. Maybe could the patch from the first post be merged (or something equivalent? It works nicely for me.

verybadsoldier, 2017-12-16:
Guys, can we reopen this issue please?

Informatic, 2019-01-13:
@verybadsoldier I don't think I am able to do it.

@slaff can we reopen this issue, considering I couldn't get it to work in the end, and @verybadsoldier still has the same issue? :)

verybadsoldier, 2019-01-13:
Yes, but I am using your fix from your original post. But I also cannot say if that is a proper way to achieve this. Big thanks for it though!

verybadsoldier, 2019-06-21:
@slaff 
Sorry for bringing this up again but this has become a critical issue for me again. I think this was not fixed in Sming (so far it was ok for me as I used the workaround @Informatic posted in the first post).

In short again: The problem is that MQTT messages (maybe other messages too but MQTT is time critical for me) only get sent with random delays at around ~500 ms. Sometimes more sometimes less.

I am in the process of switching to Sming 3.8.0 (from 3.7.0.1) and the the problem is still occuring and even the workaround does not fix it anymore on 3.8.0. I am not sure what exactly lead to the workaround not working anymore but as far as I could found out by trial-and-error it is this commit:
https://github.com/SmingHub/Sming/commit/a535785f117767bc74072a334355e267a408ddac

So, without that commit the workaround fixes the sporadic delays for me. As soon as I have that commit in I get the delays even with the workaround.

I am a bit out of luck. Would help out please?

verybadsoldier, 2019-06-22:
@slaff 
Well, some time went by and I managed to at least understand a bit of what's going on. Probably nothing new to you but sending happens in `MqttClient::onReadyToSendData`. But that function is only called periodically so buffered data which is supposed to be sent might get delayed.
Is there anything I can do about it to make MQTT messages getting sent instantly?

I came up with a dirty hack by calling `MqttClient::onReadyToSendData` manually after publish. But yeah, it's dirty and I am unsure about side effects:
https://github.com/verybadsoldier/Sming/commit/61c34e1e0bf63e2ff6c8e012201dd863acd02510

slaff, 2019-06-25:
> Is there anything I can do about it to make MQTT messages getting sent instantly?

First LWIP will call onReadyToSendData once the output buffer is not full and is ready to send more bytes. Calling it manually will not free the output buffer. But what you can do is program your mqtt client in a more "optimistic" way. So that the connection plus authentication and message MQTT packages are sent in one TCP packet and this will look like "instant" delivery. Below is a sample code.

```
mqtt = new MqttClient();
mqtt->connect(url, deviceId);
mqtt->publish(topic, message, flags);
```

verybadsoldier, 2019-06-25:
> Calling it manually will not free the output buffer. 

Hm but my workaround from above actually "fixes" the problem. There is no delay anymore. I understand your statement as that should not be the case?

> But what you can do is program your mqtt client in a more "optimistic" way. So that the connection plus authentication and message MQTT packages are sent in one TCP packet and this will look like [...]
Would I use this code everytime I want to publish a message? Would it mean to establish a new connection everytime (including the 3-way-TCP handshake)?

slaff, 2019-06-25:
> Would I use this code everytime I want to publish a message?

If the devices is not powered off and the tcp connection is still alive better just to connect once and then reuse the established tcp connection. Just call the last command over and over.

If there is a very simple example code where the issue is clearly shown and repeatable then I can take a look. Can you provide us with such a sample?

verybadsoldier, 2019-06-25:
Sure, there is no special code required. The code from the first post will surely do but also the provided `MqttClient_Hello` does the trick. I enabled some outputs and get these results:
```
39322792 Let's publish message now. Memory free=
39394072 TCP Written: 66, Available: 66, isFinished: 1, PushCount: 1 [TcpBuf: 2714]
59322795 Let's publish message now. Memory free=
59469000 TCP Written: 66, Available: 66, isFinished: 1, PushCount: 1 [TcpBuf: 2714]
79322790 Let's publish message now. Memory free=
79468964 TCP Written: 66, Available: 66, isFinished: 1, PushCount: 1 [TcpBuf: 2714]
99322790 Let's publish message now. Memory free=
99418946 TCP Written: 66, Available: 66, isFinished: 1, PushCount: 1 [TcpBuf: 2714]
119322790 Let's publish message now. Memory free=
119468975 TCP Written: 66, Available: 66, isFinished: 1, PushCount: 1 [TcpBuf: 2714]
139322790 Let's publish message now. Memory free=
139643979 TCP Written: 66, Available: 66, isFinished: 1, PushCount: 1 [TcpBuf: 2714]
159322790 Let's publish message now. Memory free=
159919008 TCP Written: 66, Available: 66, isFinished: 1, PushCount: 1 [TcpBuf: 2714]
179322792 Let's publish message now. Memory free=
179844046 TCP Written: 66, Available: 66, isFinished: 1, PushCount: 1 [TcpBuf: 2714]
199322790 Let's publish message now. Memory free=
199793989 TCP Written: 66, Available: 66, isFinished: 1, PushCount: 1 [TcpBuf: 2714]
219322790 Let's publish message now. Memory free=
219893985 TCP Written: 66, Available: 66, isFinished: 1, PushCount: 1 [TcpBuf: 2714]
239322790 Let's publish message now. Memory free=
240144035 TCP Written: 66, Available: 66, isFinished: 1, PushCount: 1 [TcpBuf: 2714]
259322790 Let's publish message now. Memory free=
259793984 TCP Written: 66, Available: 66, isFinished: 1, PushCount: 1 [TcpBuf: 2714]
279322790 Let's publish message now. Memory free=
279968985 TCP Written: 66, Available: 66, isFinished: 1, PushCount: 1 [TcpBuf: 2714]
299322790 Let's publish message now. Memory free=
300069027 TCP Written: 66, Available: 66, isFinished: 1, PushCount: 1 [TcpBuf: 2714]
319322790 Let's publish message now. Memory free=
319968982 TCP Written: 66, Available: 66, isFinished: 1, PushCount: 1 [TcpBuf: 2714]
339322790 Let's publish message now. Memory free=
339669006 TCP Written: 66, Available: 66, isFinished: 1, PushCount: 1 [TcpBuf: 2714]
359322790 Let's publish message now. Memory free=
359418992 TCP Written: 66, Available: 66, isFinished: 1, PushCount: 1 [TcpBuf: 2714]
379322790 Let's publish message now. Memory free=
380393994 TCP Written: 66, Available: 66, isFinished: 1, PushCount: 1 [TcpBuf: 2714]
```
So we see the delay between calling `publish` and the actual tcp write. Not as bad as expected actually but still jittering up to ~140 ms.

With my patch applied I get nearly instant sends:
```
45758275 Let's publish message now. Memory free=
45761582 TCP Written: 66, Available: 66, isFinished: 1, PushCount: 1 [TcpBuf: 2714]
65758271 Let's publish message now. Memory free=
65761515 TCP Written: 66, Available: 66, isFinished: 1, PushCount: 1 [TcpBuf: 2714]
85758271 Let's publish message now. Memory free=
85761505 TCP Written: 66, Available: 66, isFinished: 1, PushCount: 1 [TcpBuf: 2714]
105758268 Let's publish message now. Memory free=
105761746 TCP Written: 66, Available: 66, isFinished: 1, PushCount: 1 [TcpBuf: 2714]
125758268 Let's publish message now. Memory free=
125761775 TCP Written: 66, Available: 66, isFinished: 1, PushCount: 1 [TcpBuf: 2714]
```

verybadsoldier, 2019-06-30:
@slaff 
Did you possibly had a chance to look into this already?

verybadsoldier, 2019-07-06:
@slaff 
Could you reopen this issue please or would you prefer me to open a new one?

slaff, 2019-07-06:
Please, open a new one to restart the conversation on the latest develop code. In the new issue mention that it is related to this issue.