SmingHub/Sming #549: MQTT client does not send PING when receiving messages

Informatic, 2016-01-14:
MQTT PINGREQ when there is no outgoing activity are handled in `MqttClient::onReadyToSendData`. However, `sleep` variable checked there is reset to 0 in `TcpConnection::staticOnReceive` (as well as `onSent`). In that case `mqtt_ping` is never called when client receives multiple messages with QoS 0 with interval lower than 10 seconds.

MQTT client should store its own "sleep-like" counter variable, reset only in `onSent` event (or just use millis()?)

fcgdam, 2016-01-15:
Yes, I've also hit this issue when only receiving data, after 10s the connection is dropped.
My solution was to inform the broker by setting  keepAlive() to a large value.

hreintke, 2016-02-15:
Solve by #550

closed 2016-02-15 by hreintke