SmingHub/Sming #553: Sming future goals and priorities

anakod, 2016-01-14:
Unfortunately I have wrong mail filter for github notifications in my Gmail and didn't receive email's from @hreintke in time. On this week we have talked with @kireevco and @hreintke together. We have some ideas about future steps, but I think will be right to ask community to join our discussion.

- What is current major priority for Sming growth in your opinion?
- Which of things\features is most important?
- In which directons and parts you can take participation?

I will be glad to hear all opinions. Of course we can't do everything in same time, but more ideas - better and more complete vision. I think it's a good time for next step.

alonewolfx2, 2016-01-19:
*Major priority is ESP32 support and user friendly easy installation I think. 
I saw many users had some problems about installing sming. We need environment like arduino ide but more powerful. 
I can create startup program with nice gui  for basic installing but just on Windows unfortunately. 
Also I am trying to port all old libraries into the sming rtos version.

robotiko, 2016-01-19:
@alonewolfx2  IMHOt it will be better to joing an existing mutiplatform environment tool such as platformio or something like that, rather than develop  adhoc solutions that have to be maintained.

I think main issues when installing sming is that many times doc and install tools are not updated.

kireevco, 2016-01-19:
I agree in platformio option. They exist for about 4 years now.

hreintke, 2016-01-19:
I agree too.
There is an open issue #508.
If we want to use/support this, someone needs to get into the detail on what needs to be done and how difficult it would be to maintain.

kireevco, 2016-01-19:
I'm hoping to jump on it as soon as my day job settles down.

matoushybl, 2016-01-25:
I think that fixing bugs and documentation would be a good step forward. For example there are some problems in MQTT, some of them are already solved in PRs but not merged.

Unfortunately I don't have much time and I am also not very good at C++, but I think that I could help a bit with some usage examples.

sfam, 2016-01-29:
ESP32 and PlatformIO would be awesome!

marvinroger, 2016-02-19:
Yes, PlatformIO would be a great addition, both for Sming and for PlatformIO,where an issue already exist (platformio/platformio#398).

tprochazka, 2016-02-28:
I also vote for PlatformIO. I wanted to investigate it more, but I'm not able to found enough time neither for my Sming Smart IoT project :-(

hreintke, 2016-05-23:
Closing for now.
If further discussion needed please submit an issue in SmingRTOS.

closed 2016-05-23 by hreintke