SmingHub/Sming #555: Wrong link at install

FlintOFF, 2016-01-19:
Preblem when I try install sming using command
```
choco install sming -y -source 'https://www.myget.org/F/kireevco-chocolatey/'
```
> Downloading sming 32 bit from 'https://github.com/anakod/Sming/archive/2.1.0.zip'
Exception calling "GetResponse" with "0" argument(s): "The remote server returned an error: (404) Not Found."

Please update link to https://github.com/SmingHub/Sming/archive/2.1.1.zip

mhutchins, 2016-01-20:
HI FlintOFF

I am also affected by this issue, but I dont know enough about how choco works to be able to work around it (ahead of it being fixed). Must I wait for the fix - or can you tell me how to manually edit *something* to change the URL?

Many thanks

Martin

kireevco, 2016-01-20:
Looks like 2.1 zip archive was removed from sming archive. I will update a package to 2.4 meanwhile @hreintke will help us to revive 2.1

hreintke, 2016-01-21:
@kireevco : 
Looks like I tagges the release as 2.1.1 instead on 2.1.0.

So don't make a 2.4 but just a 2.1.1. and all will work again.

FlintOFF, 2016-01-21:
Hi @hreintke 
Still same problem
```
choco install sming.core -y -source 'https://www.myget.org/F/kireevco-chocolatey/'
```

> Attempt to get headers for https://github.com/anakod/Sming/archive/2.1.0.zip failed. Exception calling "GetResponse" with "0" argument(s): "The remote server returned an error: (404) Not Found." Downloading sming 32 bit  from 'https://github.com/anakod/Sming/archive/2.1.0.zip'

hreintke, 2016-01-21:
I updated but not sure whether it works or not.
Give it a try and feedback results.

FlintOFF, 2016-01-24:
Installation completed successfully.
Thanks.

closed 2016-01-24 by FlintOFF