SmingHub/Sming #556: Option to reconfigure http server

icodk, 2016-01-19:
Currently the HttpServer has only the addPath() method to configure the  server. I am missing the option to reconfigure. A use case could be that you allows web interface for a short while after boot, after that only one page should be available. A simple solution could  be to add clearPath method and then you could add other handlers., as before with addPath()
```
void HttpServer::clearPath(void)
{
        paths.clear();
}
```

hreintke, 2016-01-25:
@icodk : 
It is a not good idea to have only the option of a "full clear" of all paths.
If you can think of a solution to manage the paths in a generic way we can discus that here.

icodk, 2016-01-25:
@hreintke 
Can't beat your arguments. So here are mine: full clear is calling the clear() function of the HashMap existing function  so no additional complex code should be implemented. at the same time, you get full control of the web server configuration because you can again use addPath.

closed 2016-03-25 by hreintke