SmingHub/Sming #566: Windows 7 64 bit Quick Start Guide Produce Error:

sonic238, 2016-01-26:
I'm following https://github.com/SmingHub/Sming/wiki/Windows-Quickstart and got this error:

make all 
C+ app/application.cpp
In file included from c:\tools\sming\sming\smingcore\network\ntpclient.h:8:0,
                 from c:/tools/sming/Sming/SmingCore/SystemClock.h:7,
                 from c:/tools/sming/Sming/SmingCore/SmingCore.h:18,
                 from app/application.cpp:2:
c:\tools\sming\sming\smingcore\platform\station.h:36:25: error: 'SC_TYPE_ESPTOUCH_AIRKISS' was not declared in this scope
  SCT_EspTouch_AirKiss = SC_TYPE_ESPTOUCH_AIRKISS
                         ^
make: *** [out/build/app/application.o] Error 1

******************************************
maker

'SC_TYPE_ESPTOUCH_AIRKISS' was not declared in this scope

make: *** [out/build/app/application.o] Error 1

hreintke, 2016-01-26:
You are using an incompatible combination of sming and espressif sdk.
Sming 2.1.0 only works with sdk 1.4.0 and sdk 1.5.0

sonic238, 2016-01-27:
I need to fix my help .
I have no knowledge in this field.

hreintke, 2016-01-27:
Please discuss installation issues in https://gitter.im/SmingHub/Sming

closed 2016-01-27 by hreintke