SmingHub/Sming #571: Example for the ADC

ThorvaldAagaard, 2016-01-27:
I am looking for an example using the ADC with Sming, Are there any available ?

STwilight, 2016-01-30:
+1, I'm too requesting example for ADC.

hreintke, 2016-01-30:
Not much use for an application just showing `system_adc_read()`

kireevco, 2016-01-31:
@hreintke If there is such question, than probably it needs to be more verbose, maybe not in a shape of example but in a shape of QuickStart or some sort of high-level api docs.

ThorvaldAagaard, 2016-01-31:
When I write a statement like

uint16 adc = system_adc_read();

I get unresolved on the method

So a sample project giving access to the native functions in ESP8266 would be very welcom

Best regards
Thorvald

hreintke, 2016-01-31:
@ThorvaldAagaard : 
Which application/example are you using ?
When I update the basic_serial example in both master and development branch like below, it compiles OK.

```
void sayHello()
{
	Serial.print("Hello Sming! Let's do smart things.");
	Serial.print(" Time : ");
	Serial.println(micros());
	Serial.println();

	Serial.printf("This is Hello message %d \r\n", ++helloCounter);
	int a = system_adc_read();
}
```

ThorvaldAagaard, 2016-01-31:
I was using the Temperature_DS1820, but get the same error using the Basic_Serial. so it is probably something I have missing during configuration of my development environment.

I noticed that the reference to the ESP8266_SDK is just sdk/include and there are no files beneath it, might that be my problem ?

ThorvaldAagaard, 2016-01-31:
It was a setup problem as I was missing ESP_HOME as environment variable.
Thx

closed 2016-01-31 by ThorvaldAagaard