SmingHub/Sming #572: system_get_vdd33

andriyadi, 2016-01-27:
In one of my application I need to get VCC supplied to ESP board without external components. I know there's `system_get_vdd33` function, and I know that I should sacrifice ability to read ADC if using that.

But calling that function always returns 65K.

I know from [here](http://bbs.espressif.com/viewtopic.php?t=827) that in order to correctly use that function, I should: 
```The 107th byte in esp_init_data_default.bin(0～127byte) is 0xFF```

I'm not sure I understand to do that.

Please advice.

hreintke, 2016-02-15:
For reading ADC value on esp you can use : analogRead(A0)
Currently only available in develop branch.

closed 2016-02-15 by hreintke

alonewolfx2, 2016-02-15:
@hreintke its not same thing. @andriyadi it isnt about sming. its about espressif sdk. also if you use tout you cant use vdd function. we are using analogread soo we cant use vdd function on sming runtime when we use analogread.

andriyadi, 2016-02-15:
@hreintke I'm about to respond as @alonewolfx2 response.
I'm aware of using analogRead for reading analog data from external sensor/circuit. As I mentioned, I also knew that I can't use ADC input if I want to read VDD.

The solution is obvious, I need to somehow to be able to change bit of 107th of esp_init_data_default.bin automatically upon uploading.

As of now, I manually change the bit using Hex Editor program, and upload the esp_init_data_default.bin before uploading Sming-generated firmware bin. But I think there should be better solution. ESP8266 Arduino framework is managed to do that :)

Please advice. Thanks.

hreintke, 2016-02-15:
@andriyadi : @alonewolfx2 :
Sorry, I msisunderstood.

@andriyadi : Please check the arduino framework on how they implemented and feedback, best with a PR to include into Sming.

BTW : The area of esp_init_data_default is not updated when flashing Sming Software. 
Values in there stay unchaged.

andriyadi, 2016-02-15:
>BTW : The area of esp_init_data_default is not updated when flashing Sming Software. 
Values in there stay unchaged. 

Yes, I'm aware that it's not updated when flashing. That's why I changed this script `Sming/Sming/Makefile-project.mk` line 334 to:
``$(ESPTOOL) -p $(COM_PORT) -b $(COM_SPEED_ESPTOOL) write_flash $(flashimageoptions) 0x00000 $(FW_BASE)/0x00000.bin 0x7c000 $(SDK_BASE)/bin/esp_init_data_default.bin 0x09000 $(FW_BASE)/0x09000.bin``

I don't know whether it's OK. But it works so far.

As for Arduino framework, I'm still exploring it. Seems a lengthy process for my current knowledge :)

hreintke, 2016-02-15:
@andriyadi : 
The remark on  not updating is not about the file esp_init_data_default.bin.
It is the area on flash (the 0x7c000) which is not updated. One time a "make flashinit" with an updated file is enough to keep it. No need for updating the script for flashing the application.

One more remark : Beware that the 0x7c000 address is dependent on the SPI_SIZE of your esp.
The flashing to ....default.bin should than also be updated when using different SPI_SIZE.

There is a known issue that make flashinit does not take care of that different start address.
If you have the knowledge on scripts and makefiles, would be great if you can PR a solution for this.

And, please keep exploring the arduino framework. Maybe a question on that on their chat can get you in the right direction. By spreading the activities needed for improving the Sming Framework we can keep improving.

andriyadi, 2016-02-15:
> It is the area on flash (the 0x7c000) which is not updated. One time a "make flashinit" with an updated file is enough to keep it. No need for updating the script for flashing the application.

Good to know that! Thanks!

> And, please keep exploring the arduino framework...

Will do.

ZeevG, 2016-03-13:
Hi, looks like I might also be running into this issue. When using analogRead(0) I am always returned 65535.

Did you arrive at a solution or workaround?

sim0njo, 2016-06-05:
Hi,

On my standard nodeMCU, the system_get_vdd33() works fine, returning values around 3000. And system_adc_read() always returned 65535 as per documentation.

In order to use tout to measure input values and get system_adc_read() to work properly I changed the 107th byte in the esp_init area as follows:

1) determine the download address for esp_init_data_default.bin depending on flash memory size from chapter 4.2.2 on this link http://espressif.com/en/support/explore/get-started/esp8266/getting-started-guide

2) read the current 128 bytes from my nodeMCU with 4Mb flash using esptool.py:
esptool.py -p /dev/ttyUSB0 -b 115200  read_flash 0x3FC000 128 esp_init_data_default.bin

3) adjust byte at offset 107 from 0xFF into 0x1E

4) write the modified esp_init_data_default.bin back to nodeMCU with esptool.py:
esptool.py -p /dev/ttyUSB0 -b 115200  write_flash -ff 40m -fm qio -fs 32m 0x3FC000 esp_init_data_default.bin

5) reboot the nodeMCU and it works

alonewolfx2, 2016-06-05:
@sim0njo can you upload modified espinitbin to here?

sim0njo, 2016-06-05:
I can do that, but I don't know whether the other parameters will be ok for your nodeMCU. That is why I first read out the existing flash area.

Anyhow, here it is:
[esp_init_data_default.zip](https://github.com/SmingHub/Sming/files/299423/esp_init_data_default.zip)