SmingHub/Sming #574: How to access extra pins on ESP 12E modules?

ashubhatt, 2016-01-29:
Hello,

I want to access the additional pins available in ESP 12E (or in NodeMCU v3) as general GPIOs.
How can I access those pins and how to configure them as GPIOs (input or output)?
[image]

flexiti, 2016-01-29:
You can use only 2 of them,  if you open the metal casing and de-connect the port 9 and 10 from flash.memory.
(ESP will work in DIO mode and port 9 and 10 will be free)
Other uses are not yet known, but potentially could be used the same SPI for  to two devices, ie. Flash and user device on the same SPI bus

ashubhatt, 2016-01-29:
Thanks for the information. I was under the impression that these are extra pins which can be utilized as per user's will. However, upon getting some insight, I think these pins are better left alone.

closed 2016-01-29 by ashubhatt