SmingHub/Sming #580: Using HSPI

ThorvaldAagaard, 2016-01-31:
I am working on an implementation of a device placed on HSPI - I have found some nice definitions of HSPI/SPI in https://github.com/sming-framework/sming-core/tree/master/Libraries/Adafruit_ILI9341. Shouldn't these definitions be propagated to the default Sming package ? I am thinking of hspi.c hspi.c and spi_registers.h 

Long time ago we had the spi.h has that been removed ?

Thorvald

harry-boe, 2016-01-31:
We struggle as well with the current Hardware SPI implementation inside Sming.

There are currently about 3 HW SPI Implementations
-> As mentioned the one inside the IL9341 Library
-> The "official" one in "SPI.C" SPI.H" in Sming Core
-> Another one inside the "MIrf" Library (whatever that code does ? - did not found an example of this lib)

We had not really a lot of luck playing around with the HW SPI Lib from Sming-Code.
We tried it with the OLED Screen and some Other displays without luck.

Some comparison with the SPI samples from the Espresif SDK (1.4 nonos) shows some gaps wishing the Sming-Core implementation. There are as well some good blogs on the Net on how to get the HW SPI to work.

In short, I think the HW SPI lib need some rework, if not a completely new implementation.

I'm happy to support here but might need some help for testing etc.
Any opinions ?

alonewolfx2, 2016-01-31:
@harry-boe if you made new spi implementation i can test on ili9341 and nrf24l01

ThorvaldAagaard, 2016-01-31:
Looking at "SPI" in Swing-Core I found no way to change the pin-assignment, so that would be a nice feature in a new implementation.

I have some special display and will move to an official implementation of SPI if possible, and test the implementation at the same time

harry-boe, 2016-02-01:
@ThorvaldAagaard well that's intentional. The HW SPI bus is hard wired to some dedicated registers inside the esp. So no way to change them

ThorvaldAagaard, 2016-02-01:
yes, I realised that yesterday, but I had the understanding I could use the HSPI for multiple devices using different Chip select, but as far as I can see that is not possible, so we will probably switch to using SPI for one device and I2C for the other devices

avr39-ripe, 2016-02-01:
@ThorvaldAagaard No, you aren't right, you CAN use different SPI device by connecting their ChipSelect pin to different esp8266 gpio. SPI.transfer DO not touch CS pin, you can HIGH/LOW different CS pin as you need communicate with different SPI device. I do test such behaviour, it is really working :)

ThorvaldAagaard, 2016-02-01:
So when using HSPI it is stated I must use GPIO12,13,14 and 15, where GPIO15 is CS, this is fine for the first device, so what is CS for the second device ? Or do I need a totally different pin for CS for the devices and just use GPIO15 for telling HSPI there is some communication ?

avr39-ripe, 2016-02-01:
@ThorvaldAagaard with SPI class in Sming I use hardware dedicated 12,13,14 pin for CLK,MISO,MOSI (order may not correspond to pin numbers, I do not remember proper correspondence) and gpio 15 and 16 for CS for two spi devices, or gpi 4,5 for to separate cs pin.. or whatever free GPIO you have on your esp.. than when you need to communicate with some device you just properly handle devices CS pin and then doing SPI.transfer with this device, or even more - as in my case - one device is just INPUT and second - just OUTPUT so I combine SPI.transfer with two different devices at once.. look here https://github.com/avr39-ripe/HeatControl/blob/trikur/app/spi_loop.cpp#L12 for example

anakod, 2016-02-01:
I have checked hardware SPI with nrf24l01 some time before and it's working without any troubles on all bound rates.

harry-boe, 2016-02-02:
got it working with the ST7735 display as well. For whatever reason it's kind of tricky. If i'm going to change the reset pid or the CS pin its sometime not working. Maybe there is some initial device "Reset/Config" that does not really work.

fcgdam, 2016-02-10:
I'm using the standard Sming SPI functions to drive also a ST7735 display, without any issue. 
In fact comparing the way that the ili9341 SPI initialization code (for setting the SPI bus speed) and the Sming SPI functions (mainly the initialization code), it seems they are the same, so I suppose that Sming is also using HWSPI OOTB...

harry-boe, 2016-03-14:
all sorted with pr #642 
recommend to close this issue

hreintke, 2016-03-14:
As requested

closed 2016-03-14 by hreintke