SmingHub/Sming #583: integration security into SmingHub

harry-boe, 2016-02-01:
Hi all

I will soon start to support a student on a thesis about IoT security.
More specific:

The plan is to add reliable device authentication for a IoT communication scenario.
To make it reliable we plan to use a crypto chip (one of the Amtel authentication chips) for some away of authentication codes or signatures.
Obviously this requires a pairing or key exchange process along with a challenge or at least nonce/time derived token for authentication.
Next step would be to provide support for common IoT protocols (Mqtt, CoAP) to use the tokens.
First ideas is to use OpenID connect tokens to authenticate.
More scenarios might include signing Firmware images for OTA updates etc. 

Overall it might impact a lot of modules, libs etc. of Sming

So what is the best approach to get this integrated (if at all).  Some subproject, a Security folder or spread around the whole code?

Any options are welcome. Obviously any help as well ;)

PS: we already agreed with the university to build up on the esp platform ans likely simg
We also agreed on the BPM180 I2C sensor as well as one of the[ Amtel I2C Crypto Authentication ](http://www.atmel.com/products/security-ics/cryptoauthentication/) devices. They are widely available for a sub 2$ price tag.

robotiko, 2016-02-01:
@harry-boe,
it is worthy to have a look at zymbit posts  https://zymbit.com/securing-your-iot-devices-2/
They also use the new atmel device (looks like it is the only ic manufacturer in the crypto stuff ).

About Sming implications,  it is currently security unaware so they are reduced. 
I see that rBoot would have hard implications and also need a new security module for  signing/ token retrieval that would probably have some side effects in a secure layer (not in place either).
BE AWARE that  Sming currently doesn't implement any secure protocols (ssl) what invalidates any serious token usage. 
It is a relevant pending task that has not been addressed due to lack resources for it. 

It should be also considered that ESP32 comes with secure boot as specs (just in case it means something valuable).

harry-boe, 2016-02-01:
THX for the link. There are other solutions such as the FIDO crypto chips from Infineon or NXP that might be useful in this context. Problem is that i have now idea how to source them in low volumes.

> BE AWARE that Sming currently doesn't implement any secure protocols (ssl) what invalidates any serious token usage. 

i'm aware of that. That's why we need signed and maybe even encrypted tokens with a nonce or OTP to avoid reply attacks.
I also don't think that SSL might be good enough. We need end2end encryption as we cannot control the gateway in any mobile scenario

The whole subject requires a lot of thoughts .. but i see security as the key 
for the whole IoT universe

hreintke, 2016-05-23:
Initiative moved to SmingRTOS

closed 2016-05-23 by hreintke