SmingHub/Sming #585: Basic authentication and header buffer overflow

icodk, 2016-02-01:
I recently implemented Basic authentication  for the Sming web server ( I can submit an example if anybody is interested) and this exposed a possible buffer overflow issue. Basic authentication credentials are return in an "Authorization" header that is sent back from the browser after a user entered user and  password. Since there is no size limits on the fields in the browser popup login dialog  the user can enter very long strings that when parsed by the web server will  result buffer overflow( and will reset/wdt the esp).

hreintke, 2016-02-01:
@icodk : Good catch.
Example would be great. Did you update core sming also to get that implemented ?

For header buffer overflow : Can you make a PR for solving that ?

icodk, 2016-02-01:
No I didn't touch any of the SmingCore code. Just added a simple function in user space  and enabled header parsing: 
```
server.enableHeaderProcessing("Authorization");
```
Call the function before each protected page handler:
```
void onPageX(HttpRequest &request, HttpResponse &response){
         if(!authenticated(request,response)){
		    return;
	    }
.
.
      // normal response
     response.sendTemplate(tmpl); 

}
```
Authentication function:
```
bool ICACHE_FLASH_ATTR
authenticated(HttpRequest &request,HttpResponse &response){
	int i;
	
	String userPass=request.getHeader("Authorization");
	
	int headerLength=userPass.length()-6; // header in form of: "Basic MTIzNDU2OmFiY2RlZmc="so the 6 is to get to beginning of 64 encoded string
	if(headerLength>50){
		return false;
	}
	
	unsigned char decbuf[headerLength]; // buffer for the decoded string
	int outlen=base64_decode(headerLength,userPass.c_str()+6,headerLength,decbuf);
	decbuf[outlen]=0;
	userPass=String((char*)decbuf);
	if(userPass.endsWith("PASSWORD")){
		return true;
	}
	
	response.authorizationRequired() ; //authenHeader
	response.setHeader("WWW-Authenticate","Basic realm=\"MyServer\"");
	response.setHeader("401 Wrong credentials","Password required");
	response.setHeader("Connection","close");
	return false;

}
```

patrickjahns, 2016-02-21:
It would be great to include this in the HTTP Webserver core - what do you think?

icodk, 2016-02-21:
Go ahead. May be it needs to be smingified a bit. Hope you know how to do it (I don't).
If you can fix the header buffer overflow at the same time it would be great.
Let me know if I can help you.

patrickjahns, 2016-02-22:
I am not too familiar with SMING (yet) - but I`ve been looking the past days at the Http Server code finding a way to integrate Multipart formdata (and with it file upload). 

About the buffer overflow -
It would probably occur here:
https://github.com/SmingHub/Sming/blob/develop/Sming/SmingCore/Network/HttpRequest.cpp#L85
Not to sure how the buffer is "generated" but maybe we can check the current length here and if that also exceeds the allowed size stop there

Maybe post requests might also suffer from buffer overflow here? Needs some testing to be sure though
https://github.com/SmingHub/Sming/blob/develop/Sming/SmingCore/Network/HttpRequest.cpp#L195

I guess we can not assume that the client will never send more bytes than the actual content length. So we should check latest at that line if we received more than announced bytes/if the received bytes exceed the limit

icodk, 2016-02-22:
If  exceeds allocated size it is already too late. I expect buffer dynamically allocated and there we have 
to look.  In HttpRequest::parseHeader there is a limit NETWORK_MAX_HTTP_PARSING_LEN that set to 4096 HttpRequest.h However again the check is after headers data is read from the socket

patrickjahns, 2016-02-22:
I just created a hotfix - can you please check if it still persists? When testing with a large file I can`t create a bufferoverflow

patrickjahns, 2016-02-22:
One sidenote that might need to be adressed when anyone includes the authentication functionality:
Currently we send a 400 bad request when the header/post data reaches that max limit.

I noticed that google chrome would still saves the authorization header even though it was never confirmed and use it for any future request. That way it seems that any future request will fail until the cache from google is cleared. 

But for now there shouldn`t be anymore buffer overflows from header/postdata

closed 2016-04-20 by hreintke