SmingHub/Sming #587: Easy question

teicors, 2016-02-02:
Hi all,
I want to use Sming, I downloaded the "package", but using a nodemcu v3 (from Lolin) I can only blink a led (asic_Blink). I'm trying to run successfully Basic_PWM, Basic_HwPWM, but without luck. I'm so sad ...
I suppose a stupid setting, but I cannot figure it out.
Is there someone that can help me ?

Thanks in advance.

harry-boe, 2016-02-02:
What does not work. Compiling, flashing .. your sample project.
You need to be a bit more specific

teicors, 2016-02-02:
Thanks for your answer harry-boe...
I can compile, flash and run the app.. 
As example, the Basic_PWM

angelo@U36SG:~/Software/esp8266/iot/Sming-2.1.0/samples/Basic_PWM$ make flash
C+ app/application.cpp
AR out/build/app_app.a
LD out/build/app.out

#Memory / Section info:
------------------------------------------------------------------------------

out/build/app.out:     file format elf32-xtensa-le

Sections:
Idx Name          Size      VMA       LMA       File off  Algn
  0 .data         00000375  3ffe8000  3ffe8000  000000e0  2**4
                  CONTENTS, ALLOC, LOAD, DATA
  1 .rodata       000008a0  3ffe8378  3ffe8378  00000458  2**3
                  CONTENTS, ALLOC, LOAD, DATA
  2 .bss          00006548  3ffe8c18  3ffe8c18  00000cf8  2**4
                  ALLOC
  3 .irom0.text   0002fdea  40209000  40209000  000068b0  2**4
                  CONTENTS, ALLOC, LOAD, READONLY, CODE
  4 .text         00005bab  40100000  40100000  00000cf8  2**2
                  CONTENTS, ALLOC, LOAD, READONLY, CODE
------------------------------------------------------------------------------
# Generating image...
Generate firmware images successully in folder out/firmware.
Done
(!) Spiffs support disabled. Remove 'DISABLE_SPIFFS' make argument to enable spiffs.
Killing Terminal to free /dev/ttyUSB0
/home/angelo/Software/esp8266/iot/Sming-2.1.0/Sming//Makefile-project.mk:331: set di istruzioni per l'obiettivo "flash" non riuscito
make: [flash] Ucciso (ignorato)
/opt/esp-open-sdk//esptool/esptool.py -p /dev/ttyUSB0 -b 115200 write_flash -ff 40m -fm qio -fs 4m 0x00000 out/firmware/0x00000.bin 0x09000 out/firmware/0x09000.bin
Connecting...
Erasing flash...
Took 0.53s to erase flash block
Wrote 26624 bytes at 0x00000000 in 2.6 seconds (81.6 kbit/s)...
Erasing flash...
Took 1.17s to erase flash block
Wrote 196608 bytes at 0x00009000 in 19.4 seconds (81.1 kbit/s)...

Leaving...
python -m serial.tools.miniterm /dev/ttyUSB0 115200
--- Miniterm on /dev/ttyUSB0: 115200,8,N,1 ---
--- Quit: Ctrl+]  |  Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
{lSmode : null
1
2
3
4
5
6
7
8
9
10
11
.....

but the led still with the same luminosity.

harry-boe, 2016-02-02:
so that all looks all ok. I never really played around with that code.
But it might be that the initial duty change is just to small to have an effect on the LED ?

maybe you need to play around with some basePeriod setting to find a sweet spot.
check out the config method to change the settings

`void ChannelPWM::config(int duty, uint32_t basePeriod)
`

anakod, 2016-02-06:
Is you caonnec LED to right pin? May be it's different for these examples?
Try other examples also, Basic_Serial is first to try.

closed 2016-02-15 by hreintke