SmingHub/Sming #589: command handler link errors - web socket reference

harry-boe, 2016-02-04:
Tried to implement a command handler but get linker reference errors
Same is actually true the i just build the command handler sample.

`
out/build/app_app.a(application.o):(.text._Z17wsMessageReceivedR9WebSocketRK6String+0x8): undefined reference to `WebSocket::sendString(String const&)'
out/build/app_app.a(application.o): In function `_ZN15StringSumHelperC4EPKc':
/Volumes/ESPTools/harry-boe/Sming/Sming//SmingCore/Boards.h:17: undefined reference to `WebSocket::sendString(String const&)'
/Volumes/ESPTools/harry-boe/Sming/Sming//compiler/lib//libsming.a(HttpServer.o):(.text._ZN6VectorI9WebSocketED2Ev[_ZN6VectorI9WebSocketED5Ev]+0x4): undefined reference to `WebSocket::~WebSocket()'
...
`

Has there anything changed on the commend handler or web socket stuff i'm not aware off ?

hreintke, 2016-02-04:
Can you as first try rebuild the sming framework and then retry the application build ? 

Other question : are you on master or develop branch

harry-boe, 2016-02-04:
i have rebound the swing lib - was the first thing i tried
im on my own fork but in line with sming/develop

harry-boe, 2016-02-04:
it builds with swing/develop so must be my fork ?!

hreintke, 2016-02-04:
Look like it, what kind of additions you make to it ?

harry-boe, 2016-02-04:
I'm working on a Arducam - but baseline is sming/develop 

4 commits ahead of swing/develop (basically the Arducam Project 

here is my branch [https://github.com/harry-boe/Sming/tree/arducam](https://github.com/harry-boe/Sming/tree/arducam)

hreintke, 2016-02-04:
I checked out your ardocam branch and copied the application to my sample directory.
But have trouble to get it to build.(I am using eclipse ) `Info: Nothing to build for Arducam`

Will check further but will be later. 

If you find something yourself or have some bright idea on possibilities please share that here.

harry-boe, 2016-02-04:
Ok got it sorted by - had to drop my repo and recreate it

closed 2016-02-04 by harry-boe