SmingHub/Sming #590: anyone working on HW SPI support for SD Card readers ?

harry-boe, 2016-02-05:
There is only a SW SPI version which if way to slow for stuff like streaming images to SD etc.

Is someone already working on a HW SPI version ?

alainmaes, 2016-02-16:
I am interested in using hw spi too with SD card. Now I have an NRF chip hooked onto SPI which uses hw spi, but when I enable the SD card (soft SPI) together with that one nothing works anymore. Is there anybody around who can give pointers that can help me in changing the SD library to use hw spi? I won't mind to do the actual work myself.

harry-boe, 2016-02-16:
Well i struggle with the swing HW SPI lib in another context. 
I try reading bigger junks of data from a video camera module. What a get back is basically junk (random code from the stack).
Looking into the comments it appears that the initial lib was ported from a display module. Basically just writing data and not reading. 
There is one method SPI.transfer that get a buffer and a length and does magically distinguish what to do (reading or writing).

In short: I guess the HW SPI implementation is somehow incomplete. 
We are now trying to port another implementation and than try to test it with the cam and later with the SD Card.

avr39-ripe, 2016-02-16:
@alainmaes do you use separate CS (chipselect) pins for nrf and sd card??
@harry-boe sming's HW SPI is AFAIK usably complete - there is all major methods which is needed for general SPI slave to drive. spi.transfer reads and writes to spi slave in the same time (if slave support such mode of operation. If read only slave = spi.transfer whatever and you get reading result, if slave is write-only - discard what spi.transfer reads. Don\t think that some other spi implementation will solve your problem, look deeper into your camera lib.

harry-boe, 2016-02-16:
@avr39-ripe 
maybe the problem is : - if slave support such mode of operation -
We really drilled down the problem to a single SPI.transfer call and don't get any reasonable date from the camera. The same camera works fine with the Arduino/ESP implementation so it's not a camera issue.

The SPI operation is called from the TCP/IP stack with a readMemoryBlock(char* data, int bufSize);
more or less directly transformed into the SPI.transfer(data, bufSize). The buffer is allocated by the TCP/IP stack. So there is not much i can change here. 
 
here is the relevant code

`	
// called from TcpConnection::write(IDataSourceStream* stream)
// available = stream->readMemoryBlock(buffer, read);

uint16_t ArduCAMStream::readMemoryBlock(char* data, int bufSize) {
..
// get a full block
	if (len > bufSize) {
		ACAM_DEBUG("Sending Block [%d] (%d bytes)\n", bcount++, bufSize);

		SPI.transfer( (uint8_t *)data, (uint8_t)bufSize);

#ifdef ACAM_HDUMP
		hdump.print((uint8_t *)data, bufSize);
#endif
		len = len - bufSize;
		return bufSize;
	}
..
`

My next step obviously is trace it all down with a logic analyser to see where the garbage data comes from.  

Maybe you have some other ideas what might be wrong ?!

alainmaes, 2016-02-16:
@avr39-ripe You seem to have some experience with this hwspi. Could you please have a look at https://github.com/SmingHub/Sming/blob/master/Sming/Libraries/SDCard/SDCard.cpp and make an estimate how hard is it for someone not familiar with low-level hw spi to convert this? And maybe if at all possible a clue on what to replace those SDCardSPI->setMOSI() calls with. I am not asking you to do all the conversion for me, let that be clear, but if I could get a clear view on how this has to be done I will at least give it a try. I have the sd card hooked up to my ESP so I can test the result.

mralexgray, 2016-02-16:
someone should just copy what they did in `/esp8266com/libraries/SPI/SPI.h`

harry-boe, 2016-02-16:
Unfortunately it's not that easy.
The SPI stuff is widely used in many sample projects and the API has to be more or less the same as the SW SPI as the init routine is using ether one or the other implementation based on the constructor.

However i found this code sequence in the library you mentioned:

```
/**
 * Note:
 *  data need to be aligned to 32Bit
 *  or you get an Fatal exception (9)
 * @param data uint8_t *
 * @param size uint8_t  max for size is 64Byte
 * @param repeat uint32_t
 */
void SPIClass::writePattern(uint8_t * data, uint8_t size, uint32_t repeat) {
    if(size > 64) return; //max Hardware FIFO

    uint32_t byte = (size * repeat);
    uint8_t r = (64 / size);

    while(byte) {
        if(byte > 64) {
            writePattern_(data, size, r);
            byte -= 64;
        } else {
            **writePattern_(data, size, (byte / size));**
            byte = 0;
        }
    }
}

```

For whatever reason they split the data buffer into junks of 64 bytes before they actually call the SPI bus !!
If I follow the same approach i get more reasonable data. 
However there is more stuff to deal with. My device is sending a dummy bits before the data stream. (A common way to deal with latency). The ESP hw SPI bus supports such devices with register settings for dummy bits (see the Blog from [metalphreak](http://d.av.id.au/blog/esp8266-spi-dummy-bits/)) - but there is no way in the current implementation to set them !!

Bottom line there is a lot of work to get it right !!

avr39-ripe, 2016-02-17:
@alainmaes As I can see from sources it is not too difficult to port SDCard from softSPI to HW SPI. Just look in https://github.com/SmingHub/Sming/blob/master/Sming/SmingCore/SPISoft.h#L28 for  SDCardSPI->setMOSI() it is simply digitalWrite(mosiPin, VALUE) to MOSI pin.. and there so just digitalWrite(13, VALUE); than for SDCardSPI->enable = digitalWrite(15,LOW); and for *->disable = digitalWrite(15,HIGH) if you connect GPIO15 as CS (or SS) to SDCard module.. and last thing: send = writeData, and recv = readData and you must start HW SPI by SPI.begin().. try to make this substitution in current SDCard lib and see what will happen.

harry-boe, 2016-02-18:
@avr39-ripe 

I did more or less what you proposed. I changed the initialisation (creating a SPIClass instead of the SPISoft) and replaced the SPI calls and goes what. It does not work. I don't even pass the init sequence o the SD Card reader. Same HW Setup works fine with SoftwareSPI so it's not a HW issue.
I have not yet debugged the whole thing with a Logic Analyser but it's definitely not that straight forward.

The HW SPI lib seems to work if you submit single bytes but not for bigger SPI Transfers.
What i discovered from the ArduCAM with the logic analyser is lot of bytes being sent out on the bus even when you just want do do a read operation. 

What exactly have you done with the HW SPI that makes you think that the implementation is complete -> especially for longer SPI Transfers ?

avr39-ripe, 2016-02-18:
@harry-boe To be honest I _never_  try to use HW SPI with several bytes transfer.. I use it in byte sequence. :( So it seems you was _right_ it is lack of support for such mode of operation :( Had no problem to work with MCP23S17 port expanders via HW SPI, it works with 74hc165 + 74hc595 shift registers too.. It seems to work with NRF2401 via SPI. I\m not SPI guru, but afaik there is no REAL difference between byte-to-byte spi transfer and several bytes spi transfer. If I read sources of SPI and SOFTSpi the difference is whether loop is in your app to get byte-to-byte transfer of several bytes, or loop is inside spi transfer routine.

alainmaes, 2016-02-21:
Not sure if you mind changing libraries but I managed to get SD working together with NRF2401 on the same SPI bus using hw spi. You can find the code at https://github.com/WSNhub/SD
Just add that code to the Libraries folder of Sming and you can build a Sming app out of the example code in the examples folder. To be honest, the only thing I tried yet is initializing the card and showing the file system via an "ls". But if that works...

harry-boe, 2016-02-22:
@alainmaes 
Briefly liked into your code. That's basicaly a port of the Arduino SD libs including all the Implementations of the FAT filesystem and ioctl. In swing we already have them in the Services Section. 
About the Sming Hardware SPI Implementation. I spent endless nightly ours with the Logic analyser and my own new Implementation to figure out wants going on and how ESP HW SPI works.
I'm getting closer and will post my finding soon.
For a 1:1 port of the SDCard to a HW SDCard implementation i see one issue related to how SD cards handle the SPI Bus. That's basically this method:

`inline void setMOSI(uint8_t val){digitalWrite(mMOSI, val);}`

This method is basically pulling the MOSI Line hight to indicate to the SD card that data is available.
The signal will stay high or low independent of the Clock signal. 
With HW SPI this line is controlled by the HW and cannot be set with GPIO. 
I need to test if it's ok for the Card to see a 0xFF signal - interrupted by clock block start end end signals  - to be interpreted as MOSI High. 
Bottom line it needs a bit more thinking and testing for the port.

avr39-ripe, 2016-02-22:
@harry-boe I think you should change setMOSI to uint8_t readByte = SPI.transfer(0xFF); so for 8 clock periods MOSI will be HIGH and if it is needed read value is in readByte. Try this..

harry-boe, 2016-02-22:
@avr39-ripe Yep that's basically what i meant with "if it's ok for the Card to see a 0xFF signal"

I first need to finish the new HW SPI implementation. As the current one definitely does not work for more than a few bytes. Will explain later ..

harry-boe, 2016-03-05:
got that setMOSI thing sorted and ported the SDCard lib to use transfers instead of recv and send.

Was a bit tricky but is working now. I will do a PR once a added the HW SPI support

harry-boe, 2016-03-09:
HW SPI for the SD Card is include in the new SPI API PR #642

harry-boe, 2016-03-14:
implemented and merged

closed 2016-03-14 by harry-boe