SmingHub/Sming #597: Crash on digitalRead

jakakordez, 2016-02-08:
I'm trying to use an input pin for a button, so I initialized it with `pinMode(BUTTON_PIN, INPUT);`. Then I tried to read its state with `digitalRead(BUTTON_PIN)`, but the program crashed:
`Fatal exception 28(LoadProhibitedCause): epc1=0x40100779, epc2=0x00000000, epc3=0x00000000, excvaddr=0x07c00780, depc=0x00000000`
I tried on different pins and with interrupt (`attachInterrupt(BUTTON_PIN, keyPressed, GPIO_PIN_INTR_POSEDGE);`), but always same results.
My app also uses Wifi, I2C and my own SPI driver, and it works perfectly without inputs.

harry-boe, 2016-02-08:
try this link [ESP8266 Fatal exception](http://wifiesp.com/?p=7)
It will point you to the source of your problem ..

hreintke, 2016-02-08:
No need to go into disassembling.
When you compile a sming application, a firmware.map file is created. That contains the start address of all functions in the application.

Take the 0x40100779 and find the function with the "highest start address which is lower than the 0x40100779" -> That is the function in which the crash occurred.

Seeing the address it is a ROM based function from the SDK. Take a look at pointers/callbacks which are NULL when you call.

jakakordez, 2016-02-09:
Ok, so I see the problem should be with `ets_timer_disarm` function, which is at address 0x4010075c (the next one is `ets_timer_arm_new`, but its address is already 0x401007dc).

`ets_timer_disarm` is only called in `Timer::stop` function of Sming framework. After putting some `debugf` calls in Smings Timer.cpp, I found out, that the crash happens immediately after `Timer::processing` and not in mentioned stop function. So I guess disarm is called from somewhere else?

ADiea, 2016-02-09:
Or you can use the Espressif\xtensa-lx106-elf\bin\xtensa-lx106-elf-addr2line  tool

For windows:
```
C:\Espressif\xtensa-lx106-elf\bin\xtensa-lx106-elf-addr2line -f -e "<path-to-your-app>\out\build\app.out" 
```
There you can enter you address it will try to show you function and in some cases the line!

jakakordez, 2016-02-09:
Nope, there's no line number:
`0x40100779 
ets_timer_disarm
??:?`, but it correctly recognized the method. Thanks anyway. BTW I'm developing with Ubuntu.
Any other ideas?

hreintke, 2016-02-09:
Are you using a Timer in your application ?
If yes, does it have a callback function or a delegate function

jakakordez, 2016-02-09:
Yes, I am using a Timer, and I'm using it with a callback function. Should I use a delegate instead?

hreintke, 2016-02-09:
No both work but delegates can be within  an object which is already destructed -> error.

And your digitalRead is within the timer callback ?

jakakordez, 2016-02-09:
I have a digitalRead call in timers callback and one before timer initialization -> it looks like that one works fine.

hreintke, 2016-02-09:
why do you think the issue is in digitalRead ?

jakakordez, 2016-02-09:
Because my program otherwise worked fine, however I just tried to comment out the input code and I got this error:  `ets Jan  8 2013,rst cause:2, boot mode:(3,6)`. Could someone explain what does it mean (because it is different than the "Fatal exception" I got before)?

hreintke, 2016-02-09:
Reading posts in the esp8266 forum, it could be power issue

jakakordez, 2016-02-09:
Just tried to add an external 3.3V power supply, but crash persists. And it happens right on `WifiStation.waitConnection(connectOk, 20, connectFail);` call.

jakakordez, 2016-02-09:
Ok I found this: http://www.esp8266.com/viewtopic.php?p=2096#p2112 and it helped me to identify my error. It was an empty while(true) loop at the end.

hreintke, 2016-02-09:
OK, glad it is solved.

closed 2016-02-09 by hreintke