SmingHub/Sming #601: SPIFFS aligment/offset issue

kingoflolz, 2016-02-14:
I am trying to get a OTA updatable rBoot configuration with the following flash layout:

- 1 MB ROM slot 0
- 1 MB SPIFFS
- 1 MB ROM slot 1
- 1 MB SPIFFS

I am having problems programming a SPIFFS file system on to the ESP8266 and not having it get formatted by the ESP. Here are my relevant makefile flags:

    RBOOT_ENABLED ?= 1
    RBOOT_BIG_FLASH ?= 1
    RBOOT_TWO_ROMS ?= 1
    SPI_SIZE        ?= 4M
    RBOOT_ROM_0     ?= rom0
    RBOOT_LD_0      ?= rom0.ld
    RBOOT_ROM_1     ?= rom1
    RBOOT_LD_1      ?= rom1.ld
    SPIFF_SIZE      ?= 262144
    RBOOT_SPIFFS_0  ?= 0x100000
    RBOOT_SPIFFS_1  ?= 0x300000

the section of code to mount the FS

    int slot = (int) rboot_get_current_rom();
    if (slot == 0) {
        debugf("trying to mount spiffs at %x, length %d", RBOOT_SPIFFS_0 + 0x40200000, SPIFF_SIZE);
        spiffs_mount_manual(RBOOT_SPIFFS_0 + 0x40200000, SPIFF_SIZE);
    } else {
		debugf("trying to mount spiffs at %x, length %d", RBOOT_SPIFFS_1 + 0x40200000, SPIFF_SIZE);
		spiffs_mount_manual(RBOOT_SPIFFS_1 + 0x40200000, SPIFF_SIZE);
	}

and here is the command passes to esptool
`esptool.py -p /dev/ttyUSB0 -b 921600 write_flash -ff 40m -fm qio -fs 32m 0x00000 out/firmware/rboot.bin 0x02000 out/firmware/rom0.bin 0x100000 out/firmware/spiff_rom.bin`

When the ESP runs  the code, it formats the FS, clearing everything on it and prints the following debug message
`trying to mount spiffs at 40300000, length 262144`

I am very confused as the offsets should match but SPIFFS is not detecting a valid FS.

Any help will be greatly appreciated.

dseliskar, 2016-02-14:
If I understand your setup correctly RBOOT_TWO_ROMS should be 0.

raburton, 2016-02-14:
@dseliskar is correct, RBOOT_TWO_ROMS should only be set (to 1) when you intend to have two or more roms in the same memory mapped megabyte of flash. In your case your roms are in separate megabytes.

kingoflolz, 2016-02-15:
Thanks guys, The fix worked.

closed 2016-02-15 by kingoflolz