SmingHub/Sming #604: Sming SSL not compiling

crosofg, 2016-02-16:
I am unable to compile sming when SSL is enabled.  I am using esp-alt-sdk 1.4.I get the following errors.
```
CC Libraries/Adafruit_ILI9341/hspi.c
CC axtls-8266/compat/lwipr_compat.c
CC axtls-8266/replacements/mem.c
In file included from system/include/mem_manager.h:5:0,
                 from axtls-8266/replacements/mem.c:27:
axtls-8266/replacements/mem.c: In function 'ax_port_malloc':
/opt/esp-open-sdk/sdk/include/mem.h:6:27: error: implicit declaration of function 'pvPortMalloc' [-Werror=implicit-function-declaration]
 #define os_malloc(s)      pvPortMalloc(s, "", 0)
                           ^
axtls-8266/replacements/mem.c:30:16: note: in expansion of macro 'os_malloc'
 #define malloc os_malloc
                ^
axtls-8266/replacements/mem.c:41:28: note: in expansion of macro 'malloc'
     void* result = (void *)malloc(size);
                            ^
axtls-8266/replacements/mem.c: In function 'ax_port_calloc':
axtls-8266/replacements/mem.c:32:16: error: implicit declaration of function 'ets_memset' [-Werror=implicit-function-declaration]
 #define memset ets_memset
                ^
axtls-8266/replacements/mem.c:55:5: note: in expansion of macro 'memset'
     memset(result, 0, size * count);
     ^
In file included from system/include/mem_manager.h:5:0,
                 from axtls-8266/replacements/mem.c:27:
axtls-8266/replacements/mem.c: In function 'ax_port_realloc':
/opt/esp-open-sdk/sdk/include/mem.h:8:27: error: implicit declaration of function 'pvPortRealloc' [-Werror=implicit-function-declaration]
 #define os_realloc(p, s)  pvPortRealloc(p, s, "", 0)
                           ^
axtls-8266/replacements/mem.c:31:17: note: in expansion of macro 'os_realloc'
 #define realloc os_realloc
                 ^
axtls-8266/replacements/mem.c:60:28: note: in expansion of macro 'realloc'
     void* result = (void* )realloc(ptr, size);
                            ^
axtls-8266/replacements/mem.c: In function 'ax_port_free':
/opt/esp-open-sdk/sdk/include/mem.h:5:27: error: implicit declaration of function 'vPortFree' [-Werror=implicit-function-declaration]
 #define os_free(s)        vPortFree(s, "", 0)
                           ^
axtls-8266/replacements/mem.c:29:14: note: in expansion of macro 'os_free'
 #define free os_free
              ^
axtls-8266/replacements/mem.c:71:5: note: in expansion of macro 'free'
     free(ptr);
     ^
cc1: all warnings being treated as errors
make: *** [out/build/axtls-8266/replacements/mem.o] Error 1
```

slaff, 2016-02-16:
@crosofg  - I helped you today make the code compile. The changes are already in the axtls-dev branch. Can you, please, close this ticket.

crosofg, 2016-02-17:
Issue Resolved.

crosofg, 2016-02-17:
SSL not compiling in windows
```
C:\tools\Sming\Sming>make ENABLE_SSL=1
xtensa-lx106-elf-gcc -std=c99 -DESP8266 -Os -g -O2 -Wpointer-arith -Wno-implicit-function-declaration -Wl,-EL -fno-inline-functions -nostdlib -mlongcalls -mno-text-section-literals  -D__ets__ -DICACHE_FLASH -I/include -Icrypto -Issl -I.  -c -o crypto/aes.o crypto/aes.c
In file included from c:\espressif\xtensa-lx106-elf\lib\gcc\xtensa-lx106-elf\5.1.0\include\stdint.h:9:0,
                 from ssl/os_int.h:5,
                 from ssl/os_port.h:44,
                 from crypto/aes.c:38:
c:\espressif\xtensa-lx106-elf\xtensa-lx106-elf\include\stdint.h:18:19: error: missing binary operator before token "("
 #if __GNUC_PREREQ (3, 2)
                   ^
In file included from ssl/os_port.h:67:0,
                 from crypto/aes.c:38:
./util/time.h:6:3: error: unknown type name 'time_t'
   time_t tv_sec;
   ^
make[1]: *** [crypto/aes.o] Error 1
make: *** [compile-ssl] Error 2```

slaff, 2016-02-18:
@crosofg: For that issue you can try to do the following (Windows CHERTS users only)

1. Edit axtls-8266/util/time.h and change the file to look like this:
```C
#ifndef TIME_H
#define TIME_H

#include <time.h>

struct timeval
{
  time_t tv_sec;
  long   tv_usec;
};

#endif //TIME_H
```

2. Second is a "brutal-hack" that has to be done in C:\Espressif\xtensa-lx106-elf\xtensa-lx106-elf\include\stdint.h. 

Change line 18 to look like this:
```C
//#if __GNUC_PREREQ (3, 2)
#if 1
```
It would be great if someone with Windows OS can help me discover the real cause for this issue
```
c:\espressif\xtensa-lx106-elf\xtensa-lx106-elf\include\stdint.h:18:19: error: missing binary operator before token "("
 #if __GNUC_PREREQ (3, 2)
                   ^
```

hreintke, 2016-02-18:
@slaff : 
It is strange that nowhere in the whole Sming RTOS and NONOS versions (including all external libraries we include) there in any dependency on the used platform, and the axTLS implementation does.

To prevent future issues I want to keep it that way unless absolutely necessary -> exact cause and reason for not being able to solve are clear and documented.

Please take every effort in integration with the above in mind.

slaff, 2016-02-18:
@crosofg The code should compile under Windows / UDK now. Give it a try (get the latest version) and tell me if it works for you. Once it is confirmed to compile I will squash the commits in the PR to make the merge easier.

hreintke, 2016-02-18:
@slaff : 
And make the PR for RTOS :)

crosofg, 2016-02-18:
@slaff @hreintke latest version compiling successfully under windows. But a few errors on compiling Basic_SSL and MqttClient. I will raise separate issue for them.

closed 2016-02-18 by crosofg