SmingHub/Sming #612: rBoot + Spiffs Layout and persistent Setting + OTA

patrickjahns, 2016-02-22:
Hi,

i`ve been checking the documentation but have yet to figure out if the following scenario is possible:

I want to provide OTA for a community project I`ve been developing Hardware and Firmare (RGWLed Controller). There are several setting the user of the Firmware can change via webinterface and which should then persist for the future.
Is it possible to have the settings stored somewhere in flash without being overwritten by OTA?
Otherwise is it possible to keep 2 SPIFFS Filesystems in sync? 
Use case could be with the 2 rom + spiffs layout. 1 Spiff FS (A) is used by both rom slots for normal operation - the second Spiff FS (B) is only written to during update and all changes from (B) except the setting will be mirrored back to FS (A) ?

Or do you have any other suggestion?

raburton, 2016-02-23:
Sming doesn't allow you to mount two spiffs at the same time (I'm not sure if that's even possible with the underlying spiffs), so there is no way to copy files from one to another. Your use case - mixed resource files (which want updating with new roms) and config files (that should be kept between upgrades) is quite reasonable, but I'm not sure on the best way to do it. One option would be to download the files to the spiffs instead of re-flashing the whole image, that way existing files (like config) will remain.

patrickjahns, 2016-02-23:
Thank you for clarifying this.
If I configure the rom slots and roms that both would be just reading/writing to one "spiffs" block would that involve any issues? (Filesystem integrity? or anything else?)

I will use your proposed work around, but I am afraid of Man In the Middle attacks until the ssl/https client is fully operational

I am currently working on the upload feature for http post in order to avoid this - asking the user of the rom to also upload several individual files for updating it seems a bit cumbersome. (Working on a community project and I really like to provide easy updating)

raburton, 2016-02-23:
No problem using a single spiffs for two roms from the Sming point of view. But you app could create some e.g. if you update the app in one rom slot, and that affects the resources or config in the spiffs in such a way that the other rom is no longer able to function if you need to switch back to it.

patrickjahns, 2016-02-23:
Thank you for the heads up - I will integrate checks if the config version actually changed and accordingly create a backup beforehand, so it will stay compatible with old versions

Will close now here - Thank you for the help

closed 2016-02-23 by patrickjahns