SmingHub/Sming #615: Bug: missing header information (if header split between tcp packets)

patrickjahns, 2016-02-23:
While working on the HTTP_POST multipart implementation I found a glitch/bug with the current header parsing implementation

I suppose the following code in `HttpRequest::parseHeader` is meant to wait until we received the complete header:
```
int headerEnd = NetUtils::pbufFindStr(buf, "\r\n\r\n");
if (headerEnd == -1) return eHPR_Wait;
```
But what actually what might happen if the header is split between two packets, that the buffer from the first packet will be emptied from lwip.
`HttpServerConnection::onReceive` receives the `eHPR`_Wait flag and calls  `TcpConnection::onReceive(buf);` which empties the buffer from the current packet.

I think we can change this behavior if we either have a local buffer for the header, or we find a way to keep the paket inside the lwip puffer

hreintke, 2016-02-24:
@patrickjahns : 
Isn't there already a kind of buffer ?. As I understood the code it is getting header information and saving it in the request parameter hashmap. I that is the case there is no need for keeping the tcp packet, only the part of the header being processed (maybe in the same hashmap ?)

patrickjahns, 2016-02-24:
@hreintke 
Yes the code is getting header information and saving them in a Hashmap. The difficulty now is, that we first search if the header is complete by finding `\r\n\r\n`. If the first packet does not currently have the string we discard the packet and check the next - if the next packet has `\r\n\r\n` we only process that packet. 

I guess what would be also an alternative, is to process the first packet as far as we can. Keep the not processed part in a temporary buffer - then process the temporarybuffer+the next packet as far as we can etc...

I would find it easiest to not need a temporary buffer, but extend the TCP::onRCV with an optional size paramter and only clear lwip buffer until that parameter.

Assuming that calling `tcp_recved(tcp, p->tot_len)` from lwip will only clear the pbuf until the give size;
I am still not too familiar with lwip and the workings around it - maybe you can clarify on this a bit more to pin point me into a direction.

As i`ve sent you in gitter - this idea would also make it easier to implement HTTP Multipart Post (#552)

hreintke, 2016-02-24:
@patrickjahns : 
LWIP will not clear the buffer. It is done in TcpConnection::StaticOnreceive with the pbuf_free().

I am in favor of the processing of the packet as far as we can. 
The resulting header:Value combinations are now build and kept in strings.
The "temporary buffer" you are talking about can then be a string with the data up till now and which will be extended when processing the next packet. The String variable has "under the hood" a char buffer in heap and not on the stack -> will not cause stack overflow issues. This is in line with the other header processing.

For multipart & filetransfer : As far as I understand is that also send in "base64 encoded lines".
If so, shouldn't we use a similar processing for that ?

patrickjahns, 2016-02-25:
@hreintke 
If I understood you correctly - it is okay if I switch form parsing the pbuf via netutils:: methods to parsing Strings? 

Multipart/Filetransfer can but must not be send in base64 lines - will that cause a problem when we treat them as "strings" or char arrays?

patrickjahns, 2016-02-26:
@hreintke 
can you check if this is going into the direction you meant?
I tested it with several Headers and it works as expected https://github.com/patrickjahns/Sming/blob/f0820de8945968d7bed90114d8b406d81f17a6fe/Sming/SmingCore/Network/HttpRequest.cpp

If this is okay, I would start basing my multipart implementation on this. Should I seperate the pull requests for both or later squash it into one?

hreintke, 2016-02-26:
@patrickjahns : 
Looks good. 
Only thing is the code duplication in HttpRequest::extractParsingItemsList.
Did not get into the detail why you split it this way but would like to get both version to share the code.

Let's make it two PR's. When finished we get this one merged first.

patrickjahns, 2016-02-26:
@hreintke 

I left it in since parsePostData was not changed to parsing Strings instead of pbuf. But if approved I will adapter the code of parsePostData. Test the code and create a pull request. I will base the multipart version on top of that PR then

patrickjahns, 2016-04-04:
closed since #622 has been merged

closed 2016-04-04 by patrickjahns