SmingHub/Sming #616: WS2812

teicors, 2016-02-25:
Hi all,
I'm facing a strange problem playing with a strip of leds:
using the libraries "out of the box" if I light the red led, the module lights the both the red and the green led with (I suspect) the same value: this happen even if I play with the green led; but if I change the file Sming/Libraries/WS2812/WS2812.cpp commenting the lines below, where the values are swapped, the ws2812 (even if in the wrong position respect the buffer position) display the correct led and value.
I can't send here two images to fully understand my concern.

    // Rearrange R G B values to G R B order needed by WS2812 LEDs:
/*
    size_t i;
    for (i = 0; i < length; i += 3) {
        const char r = buffer[i];
        const char g = buffer[i + 1];
        buffer[i] = g;
        buffer[i + 1] = r;
    }
*/
 
Cheers,
Angelo

PS sorry for my poor English.

alonewolfx2, 2016-02-26:
Please try neopixel library and example. We will rove this ws2812 library soon.

teicors, 2016-02-28:
Hi alonewolfx2,
I changed the lib and it worked: thanks !

closed 2016-02-28 by alonewolfx2