SmingHub/Sming #620: Reverse-port / cross-compatibility for AVR (arduino-like) platforms.

mralexgray, 2016-02-26:
Taking a little ride on the `sming`-side is fun.. until you realize _you can't get off_.  🙉  Most of the "cool features" (timer stuff, etc) are implemented via `esp` only API's.  Would it be _near impossible_ to either

- maintain separate branch(es) that is/aree compatible with "other" platforms?

or, better...

- use compiler guards that use whichever API's enable `X` feature to work on `Y` device/platform? Think..

#ifdef ESP8266

Just an idea.. but it would be great to be able to move back and forth between worlds more easily - as well as be able to mix and match components from both ecosystems / maintain a single code-base.

hreintke, 2016-03-25:
Closed due to no progress

closed 2016-03-25 by hreintke

mralexgray, 2016-07-28:
Closing, unresolved... Is not progress.. It's anti-progress.  Don't be ashamed of open issues. It happens. If closed without any discussion, or resolution.... they never get fixed - and people (me) think twice about wasting their time contributing.