SmingHub/Sming #621: Hardware PWM latency on multiple analogWrites

mtavenrath, 2016-02-26:
When updating multiple HW channels using code like this
<pre>
  hwpwm.analogWrite(LED_RED, getDutyRed(red));
  hwpwm.analogWrite(LED_GREEN, getDutyGreen(green));
  hwpwm.analogWrite(LED_BLUE, getDutyBlue(blue));
</pre>

the second and third change does not get applied until another hwpwn.analogWrite is called with a slight delay. Looking at the code there seems to be a bug in espressifs if pwm_start is being called multiple times within a short time like in the the following code sequence:

<pre>
pwm_set_duty(...)
pwm_start()
pwm_set_duty(...)
pwm_start()
</pre>

I verified that calling pwm_start() only once like in this code sequence does not show the problem:

<pre>
  pwm_set_duty(getDutyRed(red), hwpwm.getChannel(LED_RED));
  pwm_set_duty(getDutyGreen(green), hwpwm.getChannel(LED_GREEN));
  pwm_set_duty(getDutyBlue(blue), hwpwm.getChannel(LED_BLUE));
  pwm_start();
</pre>

Since updating multiple PWMs at once is not an uncommon case it'd be good to have a analogWrite function which can update multiple outputs at once and works around the issue.

hreintke, 2016-02-26:
@mtavenrath 
I noticed the same with testing Sming_RTOS and implemented to circumvent :  
```
	bool analogWrite(uint8 pin, uint32 duty, bool start = true);
	bool setDuty(uint8 pin, uint32 duty, bool start = true);
```
When setting to bool start to false, the pwm_start is not executed.

I filed a bug report at espressif.

patrickjahns, 2016-02-27:
I can confirm this behavior - using in a local project in nonos a modiffied version of hardware pwm to have more fine control over it.

I set several new values and then only execute pwm_start() like this
```
pwm_set_duty(duty, 1);
pwm_set_duty(duty, 2);
pwm_set_duty(duty, 3);
pwm_start();
```

at least hardware pwm is working in sming ;-). I never could get it to run on arduino framework at all

hreintke, 2016-03-25:
Closed : Limitation of Espressif SDK

closed 2016-03-25 by hreintke

mtavenrath, 2016-03-25:
@hreintke, wouldn't it make sense to enhance the API to support updating multiple PWMs if the current API runs in a limitation of the espressif SDK?