SmingHub/Sming #623: Upgrade Arduino Json to 5.1

patrickjahns, 2016-03-01:
Arduino JSON is now available in 5.1

Changes see https://blog.benoitblanchon.fr/arduino-json-v5-1/

Interesting is also https://github.com/bblanchon/ArduinoJson/issues/134 - using floats instead of doubles. (Even though it states 5.05 - can`t find it in sming yet)

avr39-ripe, 2016-03-02:
+1 to have an option to reduce memory-hungry json encoder/decoder..

patrickjahns, 2016-03-02:
I can prepare a PR - but not sure what needs to be changed to make it compatible. 
@hreintke  does it need special attention?

hreintke, 2016-03-03:
@patrickjahns :
The ArduinoJson within is almost  the standard implementation.
Of course I do not know whether that statement will hold with the added/changed functionality from V5.1

I am much in favor of keeping in line with latest versions of used frameworks, so please PR.

patrickjahns, 2016-03-03:
@hreintke 
What does `almost` mean? ;-)

I will prepare a PR then - do we have any testcases/unittests for checking if libraries still work as expected?

patrickjahns, 2016-03-16:
Closing since merged with #627

closed 2016-03-16 by patrickjahns