SmingHub/Sming #628: MQTT sending ping

flexiti, 2016-03-03:
@Informatic 
Yours last change:
> void MqttClient::onReadyToSendData(TcpConnectionEvent sourceEvent)
>   {
>  -	-if (lastMessage and millis() - lastMessage >= 10000)
>  +	// Send PINGREQ every keepAlive time, if there is no outgoing traffic
>  +	+**if (lastMessage && (millis() - lastMessage >= keepAlive*1000))**
>   	{
>   		mqtt_ping(&broker);
>   	}

keepAlive this is the time  that is sent  to server, 
Based on this time server finds that has lost  connection with the client (client dissconeted) and this this particular time is in seconds.
Setting the same time on the server and client is hazardous.
Therefore, it should be here another time eg. new PingRepeatTime  and should be at least 2 times less down then keepAlive.
I'm not sure but it also should be less than 70 sec ( see code below)

> TcpConnection::TcpConnection(bool autoDestruct) : autoSelfDestruct(autoDestruct), sleep(0), canSend(true), **timeOut(70)**
> {
> 	initialize(tcp_new());
> }
>

Informatic, 2016-03-03:
As far as i remember MQTT specification is rather unclear on the keep alive behaviour. This is how keep alive is implemented in Paho, the most popular implementation of MQTT in Python.

flexiti, 2016-03-03:
Whatever it is,setting the same time on the server and client is hazardous.
 -  broker 10s is a litle bit shorter then client - broker sent unnesesery Last Will
 - the lack of a single ping disconnect the connection, the effect as above
You should correct it.

flexiti, 2016-03-03:
> If the Keep Alive value is non-zero and the Server does not receive a Control Packet from the Client within one and a half times the Keep Alive time period, it MUST disconnect the Network Connection to the Client as if the network had failed [MQTT-3.1.2-24].

Informatic, 2016-03-03:
...one and a half. Sending PING once every keepAlive period is fine, IMO. Actually, in my first commit I used half a period of keepAlive, but Sming maintainer opposed against that ;)
https://github.com/SmingHub/Sming/pull/550#issuecomment-171638996

flexiti, 2016-03-03:
Ok, up to You...I will close this issue.

I withdrew this change from my app, since on average every one-two hours I was disconnect and received LWT. (Mosqquito and WiFi clients -easy temporarily lose connection)
Disconnection is not yet a problem - will resume , but the receipt LWT already so :)

I really don't know why risk losing connections when only one ping is not received.
.....It's like having the entire highway, and drive on its edge. ;)

hreintke, 2016-03-07:
@flexiti @Informatic : 
in #550 I just asked the question : 
>In don't understand why we would send PING at a rate of keepAlive/2. Shouldn't that just be keepAlive ?

Answer : 
>as it seems to be behaviour of python-paho MQTT implementation, which seems to work fine.

But now I understand that the current implementation has a design issue : 
>Whatever it is,setting the same time on the server and client is hazardous.

That is a valid statement,  and the current implementation should be changed.
If one of you can create a PR for this I will merge in both Sming and SmingRTOS.

flexiti, 2016-03-07:
In my opinion they should be separated times.
KeepAlive  can tell us that system(broker)  los connections, eg. an kind of alarm from broker to the other user using LWT,
and the typical value will be from several seconds to several minutes.
PingTime is usually from a few to tens of seconds (it is nature of tcpconnection).
(Of course, always PingTime must be smaller or  equal to the KeepAlive :) )
I would suggest creating variable eg. PingTime with a default value, eg. 10 seconds, and KeepAlive (already exist) with a default value of 1 minute.
(I use such times, the system works flawlessly 24 hours a day)
Additionally, the method SetPingTime should be added, so everyone will be able to adjust both the times to meet their needs.
If it is a good solution I can create PR.

flexiti, 2016-03-14:
#653 added

closed 2016-03-14 by flexiti