SmingHub/Sming #630: Problems in Basic_AirUpdate example

braindamagedman, 2016-03-04:
When i try update firmware over the air in 99.99% i get the error:

`Download file:
  (0) http://192.168.1.3/esp/firmware/0x00000.bin -> 0x0
Download: http://192.168.1.3:80/esp/firmware/0x00000.bin
connect to: 192.168.1.3
TcpConnection::connect result:, 0
realloc 59 -> 190
TCP connection error: -8
 
Firmware download failed..`

When i download file in browser it's work correct. I try to ping my ESP module - sometimes ping packets are lost. And i see in debug output that wifi connection was reset.

When i am lucky and firmware downloads correctly i get another problem:

`Download file:
    (0) http://192.168.1.3/esp/firmware/0x00000.bin -> 0x0
Download: http://192.168.1.3:80/esp/firmware/0x00000.bin
connect to: 192.168.1.3
TcpConnection::connect result:, 0
realloc 59 -> 190
OnConnected
TCP connected
onReadyToSendData: 0
TcpClient request completed
TCP sent: 61
onReadyToSendData: 2
Header pos: 246
Server === nginx/1.8.1
Date === Tue, 01 Mar 2016 21:02:24 GMT
Content-Type === application/octet-stream
Content-Length === 31392
Last-Modified === Tue, 01 Mar 2016 20:19:04 GMT
Connection === close
ETag === "56d5f938-7aa0"
Accept-Ranges === bytes
TCP received: 1460 bytes
...
...
TCP received: 982 bytes
onReadyToSendData: 1
TCP received: (null)
TCP connection closing
-TCP connection
jump: 5472
Download file:
    (1) http://192.168.1.3/esp/firmware/0x09000.bin -> 0x9000
Download: http://192.168.1.3:80/esp/firmware/0x09000.bin
+TCP connection
connect to: 192.168.1.3
TcpConnection::connect result:, 0
realloc 59 -> 190
OnConnected
TCP connected
onReadyToSendData: 0
TcpClient request completed
TCP sent: 61
onReadyToSendData: 2
Header pos: 248
Server === nginx/1.8.1
Date === Tue, 01 Mar 2016 21:02:24 GMT
Content-Type === application/octet-stream
Content-Length === 225928
Last-Modified === Tue, 01 Mar 2016 20:19:04 GMT
Connection === close
ETag === "56d5f938-37288"
Accept-Ranges === bytes
TCP received: 1460 bytes
onReadyToSendData: 1
[... 219 lines cut ...]
bool state = true;
 
void blink()
{
                digitalWrite(LED_PIN, state);
        state = !state;
}
 
void init()
{
    spiffs_mount(); // Mount file system, in order to work with files
 
    Serial.begin(SERIAL_BAUD_RATE); // 115200 by default
    Serial.systemDebugOutput(true); // Debug output to serial
 
    WifiStation.enable(true);
    WifiStation.config(WIFI_SSID, WIFI_PWD);
    WifiAccessPoint.enable(false);
 
    // Run our method when station was connected to AP
    WifiStation.waitConnection(connectOk);
 
    pinMode(LED_PIN, OUTPUT);
        procTimer.initializeMs(100, blink).start();
}`

alonewolfx2, 2016-03-04:
Can you try basic_rboot. we are using rboot for ota update. basic airupdate not working correctly anymore due to flash size support. we will remove this example soon.

braindamagedman, 2016-03-05:
Ok, i try to use basic_rboot. This is my code [](http://paste.fedoraproject.org/334407/18383414/)

When i type ota in serial session i get error:

`Updating...
~TCP connection
+TCP connection
Download file:
    (0) http://192.168.1.3/esp/firmware/rom0.bin -> 82000
Download: http://192.168.1.3:80/esp/firmware/rom0.bin
connect to: 192.168.1.3
TcpConnection::connect result:, 0
realloc 56 -> 187
OnConnected
TCP connected
onReadyToSendData: 0
TcpClient request completed
TCP sent: 58
onReadyToSendData: 2
Header pos: 248
Server === nginx/1.8.1
Date === Sat, 05 Mar 2016 13:14:09 GMT
Content-Type === application/octet-stream
Content-Length === 260000
Last-Modified === Sat, 05 Mar 2016 12:28:35 GMT
Connection === close
ETag === "56dad0f3-3f7a0"
Accept-Ranges === bytes
TCP received: 1460 bytes
onReadyToSendData: 1
TCP received: 1460 bytes
onReadyToSendData: 1
TCP received: 1460 bytes
onReadyToSendData: 1
TCP received: 1460 bytes
onReadyToSendData: 1
TCP received: 1460 bytes
onReadyToSendData: 1

 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset`

hreintke, 2016-03-25:
Closed due to no progress

closed 2016-03-25 by hreintke

braindamagedman, 2016-11-16:
I think it was power supply problem. After i stopped to use socket breadboard problem disappeared.