SmingHub/Sming #632: SDK 1.5.0 produce bigger image then 1.3.0, cause issue when downloading

icodk, 2016-03-04:
Try to move from sdk 1.3.0 to 1.5.0, but image size. is too big and spiff_bin is overwriting 0x0A000.bin 
Here are the numbers for 1.3.0 :
```
#Memory / Section info:
------------------------------------------------------------------------------
   Section|                   Description| Start (hex)|   End (hex)|Used space
------------------------------------------------------------------------------
      data|        Initialized Data (RAM)|    3FFE8000|    3FFE8374|     884 (-44)
    rodata|           ReadOnly Data (RAM)|    3FFE8380|    3FFEBE3C|   15036 (-312)
       bss|      Uninitialized Data (RAM)|    3FFEBE40|    3FFF36F8|   30904 (+1776)
      text|            Cached Code (IRAM)|    40100000|    4010575A|   22362 (-1296)
irom0_text|           Uncached Code (SPI)|    4020A000|    402532D2|  299730 (-19840)
Total Used RAM : 46824
Free RAM : 35096
Free IRam : 10424
------------------------------------------------------------------------------
# Generating image...
   Section|                   Description| Start (hex)|   End (hex)|Used space
------------------------------------------------------------------------------
      data|        Initialized Data (RAM)|    3FFE8000|    3FFE8374|     884
    rodata|           ReadOnly Data (RAM)|    3FFE8380|    3FFEBE3C|   15036
       bss|      Uninitialized Data (RAM)|    3FFEBE40|    3FFF36F8|   30904
      text|          Uncached Code (IRAM)|    40100000|    4010575A|   22362
irom0_text|             Cached Code (SPI)|    4020A000|    402532D2|  299730
------------------------------------------------------------------------------
Entry Point : 40100004 call_user_start()
Total Used RAM : 46824
Free RAM : 35096
Free IRam : 10406 or 26790 if 48k IRam
Generate firmware images successully in folder out/firmware.
Done
# Generating spiffs_bin
Checking for spiffs files
web/dev directory exists. Creating out/firmware/spiff_rom.bin
Creating rom 'out/firmware/spiff_rom.bin' of size 0x80000 (524288) bytes.
.
.
out/firmware/spiff_rom.bin---------->0x54000
```
And the download command:
```
/c/Espressif/utils/esptool.exe -p COM5 -b 256000 write_flash -ff 40m -fm qio -fs 32m 0x00000 out/firmware/0x00000.bin 0xA000 out/firmware/0x0A000.bin    0x54000 out/firmware/spiff_rom.bin
```
No conflict in download (spiffs_bin starts after 0x0A000.bin)
```
Writing at 0x00052c00... (99 %) 
Writing at 0x00053000... (100 %) 
Written 300032 bytes in 13.93 seconds (172.28 kbit/s)...
Erasing flash...
head:  12 ;total: 128
erase size :  475136

Writing at 0x00054000... (0 %) 
```

Here are the numbers for 1.5.0 :
```
#Memory / Section info:
------------------------------------------------------------------------------
   Section|                   Description| Start (hex)|   End (hex)|Used space
------------------------------------------------------------------------------
      data|        Initialized Data (RAM)|    3FFE8000|    3FFE83A0|     928 (+44)
    rodata|           ReadOnly Data (RAM)|    3FFE83A0|    3FFEBF94|   15348 (+312)
       bss|      Uninitialized Data (RAM)|    3FFEBF98|    3FFF3160|   29128 (-1776)
      text|            Cached Code (IRAM)|    40100000|    40105C6A|   23658 (+1296)
irom0_text|           Uncached Code (SPI)|    4020A000|    40258052|  319570 (+19840)
Total Used RAM : 45404
Free RAM : 36516
Free IRam : 9128
------------------------------------------------------------------------------
# Generating image...
   Section|                   Description| Start (hex)|   End (hex)|Used space
------------------------------------------------------------------------------
      data|        Initialized Data (RAM)|    3FFE8000|    3FFE83A0|     928
    rodata|           ReadOnly Data (RAM)|    3FFE83A0|    3FFEBF94|   15348
       bss|      Uninitialized Data (RAM)|    3FFEBF98|    3FFF3160|   29128
      text|          Uncached Code (IRAM)|    40100000|    40105C6A|   23658
irom0_text|             Cached Code (SPI)|    4020A000|    40258052|  319570
------------------------------------------------------------------------------
Entry Point : 40100004 call_user_start()
Total Used RAM : 45404
Free RAM : 36516
Free IRam : 9110 or 25494 if 48k IRam
Generate firmware images successully in folder out/firmware.
Done
# Generating spiffs_bin
Checking for spiffs files
web/dev directory exists. Creating out/firmware/spiff_rom.bin
Creating rom 'out/firmware/spiff_rom.bin' of size 0x80000 (524288) bytes.
.
.
out/firmware/spiff_rom.bin---------->0x58000
```
And the download command for 1.5.0:
```
/c/Espressif/utils/esptool.exe -p COM5 -b 256000 write_flash -ff 40m -fm qio -fs 32m 0x00000 out/firmware/0x00000.bin 0xA000 out/firmware/0x0A000.bin    0x58000 out/firmware/spiff_rom.bin
```
And the conflict in download (spiffs_bin starts at the same place the last chank from  0x0A000.bin):
```
Writing at 0x00057c00... (99 %) 
Writing at 0x00058000... (100 %) 
Written 320512 bytes in 14.85 seconds (172.63 kbit/s)...
Erasing flash...
head:  8 ;total: 128
erase size :  491520

Writing at 0x00058000... (0 %) 
Writing at 0x00058400... (0 %) 
```

hreintke, 2016-03-08:
@icodk : 
I have seen the same when moving to SmingRTOS. 
Will check whether how/if the patch can be backported.
It is dependent on exact application size (boundary error)  , so small updates/extensions to applications can prevent the issue.

icodk, 2016-03-08:
@hreintke :
Do you refers to the size issue or the wrong  spiff start calculation(in this case  0x00058000) ?
I am using 4MB esp-12 so I set SPIFF_START_OFFSET = 0x100000 and use a 
```
spiffs_mount_manual(0x100000+ 0x40200000,  0x80000 );
```
This solved the spiff start calculations issue.
Hope you could answer the following questions:
1. Maximum program size on a 4MB flash (ESP-12)  is about 1MB (1048576  - 16384 - 36864 )?
2. Program size  is given by the line :
```
irom0_text|             Cached Code (SPI)|    4020A000|    40258052|  319570
```
about 32% (319570/(1048576- 16384 - 36864) of the max program space?
3. To make an image that will be loaded on the begining of the third 1MB boundery (0x200000) I should modify the link script from:
```
irom0_0_seg :                         org = 0x4020A000, len = 0xF0000
```
to:
```
irom0_0_seg :                         org = 0x60000000, len = 0xF0000
```
?
Many thanks in advance

hreintke, 2016-03-09:
@icodk : 
Yes I referred to the 0x58000. It is the calculation of the first free sector after the application which fails on a boundary issue (both in makefile & framework). 
By setting the SPIFF_START_OFFSET = 0x100000 & mount_manual you circumvent the issue.
1/ Max Application size is 1MB, limitation of the cpu/architecture. 
2/ `irom0_text` : that is only part of the program size, I do not recall all the details now.
3/image
The `irom0_0_seg :                         org = 0x4020A000, len = 0xF0000` tells the Linker at what relative location the application will be flashed. The len should be dependent on SPI_SIZE but that is not (yet ?) implemented in Sming. (Need some makefile magic, so if have that knowledge :smile: ).

The 0x4020A000 has two parts 1/ 0x40200000 -> The memory address by which the flash will be accessed. 2/ 0xA000 -> The offset.
The  0x4020A000 is the same when flashing the application on any 1MB boundary.
The actual flash location is First MB 0x0000000 second MB 0x100000 third MB 0x200000.....

So for application in third MB -> Linker 0x4020A000, flash location 0x20A000

But.. also the boot needs to know where the application to run is. That is handled by configuring rBoot.

icodk, 2016-03-09:
@hreintke:
Thanks for the detailed answer. Just to be sure I understand: The actual addresses to set in the linker script are(?): 0x4020A000 for the first MB, 0x5020A000 for the second,0x6020A000 for the thired.

hreintke, 2016-03-09:
No, In the linker file it is always 0x4020A000. 

In the boot process the physical location of the application is always mapped to 0x4020A000.

SamuelGold, 2016-03-12:
SDK have 0x09400 size, but firmware writes to 0x09000, so it overwrite SDK and program does not work. How to fix it?

SamuelGold, 2016-03-21:
To fix do this:
1. Add IMAGE_SDK	= 0x0A000.bin to Makefile-user.mk in your project home.
2. Replace all 9000 in ./Sming/Makefile-project.mk to A000
3. Open ./Sming/compiler/ld/eagle.app.v6.cpp.ld and set irom0_0_seg:
   irom0_0_seg :                         org = 0x4020A000, len = 0x78000

And then works!

hreintke, 2016-03-21:
@SamuelGold : 
That is a solution for another issue.
When booting fails with csum err -> use the one you describe

closed 2016-03-21 by hreintke