SmingHub/Sming #634: spiffs overwrite bootloader

braindamagedman, 2016-03-06:
I try to compile [this code](http://paste.fedoraproject.org/334782/72601831)

When i set DISABLE_SPIFFS = 1 in Makefile all works fine.
But when i am activate SPIFFS my ESP doesn't start.

I see this in log:
esptool.py -p /dev/ttyUSB1 -b 115200 write_flash -ff 40m -fm qio -fs 8m 0x00000 /usr/share/nginx/esp/firmware/rboot.bin 0x02000 /usr/share/nginx/esp/firmware/rom0.bin 0x100000 /usr/share/nginx/esp/firmware/spiff_rom.bin

Why spiffs offset is 0x100000??
I have only 1mb flash: SPI Flash Size: 1048576 - and 0x100000 actually is 1048576 in decimal.

When i dump flash content - esptool.py --port /dev/ttyUSB1 --baud 115200 read_flash 0x00000000 2320  - i get this in binary:

00000000  01 80 01 00 ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
00000010  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
*
000000f0  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff 00 00  |................|
00000100  01 80 00 00 f8 00 00 00  28 00 00 00 01 74 65 73  |........(....tes|
00000110  74 66 69 6c 65 2e 74 78  74 00 00 00 00 00 00 00  |tfile.txt.......|
00000120  00 00 00 00 00 00 00 00  00 00 00 00 00 00 02 00  |................|
00000130  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
*
00000200  01 00 00 00 fc 54 68 69  73 20 6c 69 6e 65 20 69  |.....This line i|
00000210  73 20 74 68 65 20 63 6f  6e 74 65 6e 74 20 6f 66  |s the content of|
00000220  20 74 65 73 74 20 66 69  6c 65 2e 0a 0a ff ff ff  | test file......|
00000230  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
*
00000910

patrickjahns, 2016-03-06:
See the wiki regarding rBoot and spiffs here; https://github.com/SmingHub/Sming/wiki/rBoot-and-OTA-updating

PTDreamer, 2016-03-08:
I think that the DISABLE_SPIFFS flag usage is ambiguous. Not having looked too hard it looks like the flag only controls if the binary blob containing files to be put on flash is uploaded or not.
In terms of it actually disabling the file system I can't see how/where that is done.
The offset value itself is defined on the makefiles and it is independent of the flash size. It can be controlled with the "RBOOT_SPIFFS_0=new_size" compile flag.

hreintke, 2016-03-08:
@PTDreamer : 
The DISABLE_SPIFFS is controlling whether a spiffs_rom.bin is being created and flashed to the esp.
Whether the application does or does not mount the spiffs fs is up to the application by using spiffs_mount or spiffs_mount_manual -> independent on the setting of DISABLE_SPIFFS.

The use-cases for this are : 
- Don't flash a spiffs fs when not needed by application
- Don't flash a spiffs fs when the application want to mount an existing spiffs fs which is already on the esp.

PTDreamer, 2016-03-09:
> @PTDreamer :
> The DISABLE_SPIFFS is controlling whether a spiffs_rom.bin is being created and flashed to the esp.

That was why I said the naming is ambiguous, it is called "DISABLE_SPIFFS" (disable....file system) when in fact you are not disabling it, you are only preventing a new FS binary blob to be flashed.

braindamagedman, 2016-03-09:
I solve my problem by adjusting size and location of spiffs (RBOOT_SPIFFS_0, RBOOT_SPIFFS_1 and SPIFFS_SIZE). I just wonder that writing out of upper limit of flash overwrites bootloader. Is it normal behavior?

hreintke, 2016-03-25:
Works as intended

closed 2016-03-25 by hreintke