SmingHub/Sming #643: httpClient needs (?) some delay between subsequent requests

avr39-ripe, 2016-03-09:
while coding my app found that there is a need for delay (tested minimum 100ms) between subsequent httpClient.downloadString calls. If there is NO delay and requests goes one after another - after first successful request I get TcpClient error -15 (from memory, not real copy-paste, but -15 is the precise number!) delays implemented on timers. Server for that test was ANOTHER ESP8266 with sming hosted webserver. Do not test this issue with "big scary webserver" yet. Just to inform community and because mr. @patrickjahns insist to open issue :) May be someone else can add something to this.
meta code:
loopTimer starts first request - than in onCompleted we process response and arm httpDelayTimer to wait 100ms and then again call function which make second request and so on in a loop,, when there was no httpDelayTimer second and all following requests fail with error -15.

patrickjahns, 2016-03-12:
@avr39-ripe 

Did you experiment further with this? I tried but couldn`t replicate your problem when using a normal webserver (nginx)

avr39-ripe, 2016-03-12:
no further experiments yet. may be just sming httpClient - to - sming httpServer issue. try to find some time to check it from httpServer side...

patrickjahns, 2016-03-12:
I think your issue can be related to 
#456

It would be great if you enable debugging via `Serial.systemDebugOutput(true)`; and provide a more verbose output of your requests to trace down the error.

patrickjahns, 2016-03-14:
@avr39-ripe 
Can you provide a verbose log on your findings? (debug messages enabled)

patrickjahns, 2016-03-15:
@avr39-ripe 

Can you please provide the code you experienced the issue with?

avr39-ripe, 2016-03-15:
https://github.com/avr39-ripe/TyTherm/blob/c8bdbd7557016236d9991f8d8e4dff05f97def51/lib/tempsensor/tempsensors.cpp#L192
This part of code do not work properly,
https://github.com/avr39-ripe/TyTherm/blob/multisensor_client_app/lib/tempsensor/tempsensors.cpp#L182 and this code starts to work as expected

patrickjahns, 2016-03-16:
Looking at your code I can confirm my assumptions. Essentially it is the same as #456 - I propose to close this and move discussion/resolution to the other issue since it has more information already

avr39-ripe, 2016-03-16:
Resolution of this issue moved to #456

closed 2016-03-16 by avr39-ripe