SmingHub/Sming #644: RFID-RC522

teicors, 2016-03-10:
Hi all,
I woud use the rfid module, but when I compile Sming, I get a lot of error: is there any direction to compile the library ?

I'm usnig Sming 2.1.0 and I put the rfid files (git clone https://github.com/miguelbalboa/rfid.git) into the Library directory. Launching make i have this errors:

angelo@U36SG:~/Software/esp8266/iot/Sming/Sming$ make
C+ Libraries/rfid/MFRC522.cpp
In file included from Libraries/rfid/MFRC522.cpp:10:0:
Libraries/rfid/MFRC522.h:384:15: error: '__FlashStringHelper' does not name a type
  static const __FlashStringHelper *GetStatusCodeName(StatusCode code);
               ^
Libraries/rfid/MFRC522.h:388:15: error: '__FlashStringHelper' does not name a type
  static const __FlashStringHelper *PICC_GetTypeName(PICC_Type type);
               ^
Libraries/rfid/MFRC522.cpp: In member function 'void MFRC522::PCD_WriteRegister(byte, byte)':
Libraries/rfid/MFRC522.cpp:43:6: error: 'class SPIClass' has no member named 'beginTransaction'
  SPI.beginTransaction(SPISettings(SPI_CLOCK_DIV4, MSBFIRST, SPI_MODE0)); // Set the settings to work with SPI bus
Libraries/rfid/MFRC522.cpp:43:35: error: 'SPI_CLOCK_DIV4' was not declared in this scope
  SPI.beginTransaction(SPISettings(SPI_CLOCK_DIV4, MSBFIRST, SPI_MODE0)); // Set the settings to work with SPI bus
[....]

I reported here only the fist lines.
I'm neither a great programmer in c++ nor deep in Arduino env. but looking at the code of Adafruit_ST7735.cpp, I can found both the SPI.beginTransaction and the __FlashStringHelper.

Thanks for any kind help !!

alonewolfx2, 2016-03-10:
> is there any direction to compile the library ?

just wait for a while. i ported rfid library and i will upload wen its cleaned. :)

teicors, 2016-03-11:
Thanks alonewolfx2 !! :)
Please feel free to contact me if you need some help for debugging !! ;)

alonewolfx2, 2016-03-11:
please contact me in gitter channel [https://gitter.im/SmingHub/Sming](url)

teicors, 2016-03-11:
I receive an error 404 clicking on the link

teicors, 2016-03-11:
i'm in

hreintke, 2016-03-25:
Closed due to no progress

closed 2016-03-25 by hreintke