SmingHub/Sming #645: GET request to control a GoPRO with Sming

silvergaucho, 2016-03-10:
Hei guys,

i'm just playing with Sming and wanted to control my GoPro Hero 3. I can connect to it
but I cannot forward messages.

I've founded the gopro commands and I'm implementing this:

HttpClient goPro as main object

than after connectionOK

goPro.downloadString("http://10.5.5.9/bacpac/SH?t=MYPASSWORD&p=%01",onDataSent);

but nothing happens. The string is correct and if I paste it on Chrome I can take a shot.
What I'm doing wrong?

Sorry if it sounds stupid, maybe I need to get more into Sming.

Thank you all.

patrickjahns, 2016-03-16:
Please describe what you meant by "nothing happens".
Please also enable debug messages and provide a verbose log

hreintke, 2016-03-25:
Incomplete request

closed 2016-03-25 by hreintke