SmingHub/Sming #648: TCP Already freed message

hreintke, 2016-03-12:
@ADiea : 
I know that in a comment to one of the issues/PR's you mentioned the solution for the "already freed" message. I cannot find that anymore.
Can you repeat your solution and/or submit a PR to solve ?

ADiea, 2016-03-13:
TCPConnection::close() can be called from the outside in which case checkSelfFree needs to be called to actually remove the TCPConnection from heap
But the destructor also calls close()

solution is to modify destructor to clear autoSelfDestruct prior to calling close() Like this
```
TcpConnection::~TcpConnection()
{
    autoSelfDestruct = false;
    close();
    debugf("~TCP connection");
}
```

will make a pr soon

piperpilot, 2016-03-16:
This needs to be fixed ASAP please.  the original reporting was in the following PR:

https://github.com/SmingHub/Sming/pull/514

I can confirm that the above fix does get rid of the "already freed" message.

hreintke, 2016-03-16:
@piperpilot : 
Fastest way to get an issue solved is creating a PR for merging. ;)

piperpilot, 2016-03-17:
I'd be happy to make a PR with the above change, but since I am not the subject matter expert, if there is something wrong, I probably can't fix it.  I'll go ahead and submit a PR though so we can get this change through.

closed 2016-03-17 by hreintke

hreintke, 2016-03-17:
@piperpilot : 
You already verified that the change was working, as did @Adiea.

Instead of me making all PR's for bugs found by all, I prefer to get PR's with the solution from the community. That saves the time that I then can spend on my own little project which I also enjoy working on.

Thanks for making the PR. Merged.

piperpilot, 2016-03-17:
No problem @hreintke ...I was hoping the original committer would fix the issue since it was a regression, but I'm happy to help.

I'm not trying to figure out what changed with wifiStation and wifiAP as there are some behavior differences.