SmingHub/Sming #649: Improve documentation / ease of entry

patrickjahns, 2016-03-12:
The functionality of SMING is great - but at some points it lacks some clear structure in regards to documentation. At the beginning it is still hard to find where the documentation resides (i.e. sminghub.github.io)

- promote sminghub.github.io more prominently on the front page
- cleanup wiki (i.e. example wiki page) 
- cleanup examples (maybe even enforce a README per example? )

icodk, 2016-03-12:
+100
To the 3 point from @patrickjahns  I would add  the following:
A general explanation/instruction about how this "form" is working. As far as I can see, it is a mix of development and support forum. Is it acceptable to ask support questions?
Another challenge is  to get to understand how development/contribution is done, for somebody that is not too familiar with github ecosystem (like me), Who is in charge ? **cloning** a branch to a private **repository,** adding your code, submitting a **pull request**, **merging** , UNI...
Those that can describe are also best for the actual development. This will make this masterpiece in collaboration even greater.

patrickjahns, 2016-03-12:
@icodk 

In regards to Contributions there is already a [straight forward notice](CONTRIBUTING.md)
It also outlines the flow for the project. For general github workflow see the many tutorials out there 

The project maintainers can be referred to has the organization (SmingHub) members - but clarifying this in a README is a nice addition

icodk, 2016-03-12:
straight forward notice is broken

patrickjahns, 2016-03-12:
Forwhatever reason relative links didn`t work - here is the full link:
https://github.com/SmingHub/Sming/blob/develop/CONTRIBUTING.md

hreintke, 2016-05-23:
Issue need to be moved to SmingRTOS.

closed 2016-05-23 by hreintke