SmingHub/Sming #650: Update examples

patrickjahns, 2016-03-12:
In gitter @avr39-ripe mentioned that some examples are "out dated" and use the framework in a "wrong way".

Let`s track the outdated examples in this issue and update them accordingly

piperpilot, 2016-03-16:
httpServer_ConfigNetwork seems to be broken, or at least it doesn't work well like it used to.  When you connect now, the spinner just spins.  It seems that the accesspoint disconnects the client, or something and you never get an update.  This example is probably our best "starting point" for users wanting to create a project.  We should probably fix it.

hreintke, 2016-05-23:
We will review the (working of) examples within SmingRTOS.
Çlosing for now.

closed 2016-05-23 by hreintke