SmingHub/Sming #651: HttpServer only allowing for one cookie value

patrickjahns, 2016-03-12:
I just found out that `HttpResponse::setCookie(const String name, const String value)` only allows for one cookie value. This is limiting cookie responses

I see two options to go be more consistent with HTTP standards:
1) setcookie utilizes an internal String buffer and appends `key=value`; to the cookie String
2) change the interface to `HttpResponse::setCookie(const String cookie)` and let the user create a valid cookie string

Matzz, 2020-12-07:
I think it would be good to also add `getCookie` / `getCookies` methods to HttpRequest

slaff, 2021-03-12:
> I think it would be good to also add getCookie / getCookies methods to HttpRequest

@Matzz can you propose a PR with the changes?

Matzz, 2021-03-12:
@slaff (cc @mikee47) I was tempting to do so but its turn out that is much complex than I though. According to HTTP standard Response cookies are set by providing the same `Set-Cookie` header multiple times.

However as you could see in the following code:
```
HttpResponse* HttpResponse::setCookie(const String& name, const String& value)
{
	String s = name;
	s += '=';
	s += value;
	headers[HTTP_HEADER_SET_COOKIE] = s;
	return this;
}
```
Sming support only one header per key. I don't have any good idea how to implement that. Maybe by adding separate field in the `Response` called cookies and which will be a subset of headers?

Another problem is how much of the spec we would like to support here? Cookies are very complex area in HTTP spec. We have all those security modification and cookies set per paths. IMHO we could provide a way to set those modificators but without any helpers/validation. Just Map of modificator -> value pairs.

Regarding the `Request` side its easiers. Cookies are separated by `;`. Probably would just need to add some way handling escaped semicolon (not sure what spec says about that)

Reference:
https://developer.mozilla.org/en-US/docs/Web/HTTP/Headers/Set-Cookie
https://developer.mozilla.org/en-US/docs/Web/HTTP/Headers/Cookie

slaff, 2021-03-26:
A PR adding support for setting multiple cookie is on its way. See #2290.

slaff, 2021-03-29:
PR #2290 is merged in the `develop` branch and setting multiple cookies should be possible.

closed 2021-03-29 by slaff