SmingHub/Sming #652: HttpServer_ConfigNetwork example not updating network list

mxaris, 2016-03-13:
As i can see in the example the WifiStation.startScan() method called only one time in init() so the onAjaxNetworkList() always returns the same data. How to change the onAjaxNetworkList to perform a new scan every time its called.

patrickjahns, 2016-03-14:
Scanning for Networks and having the results in a list is not synchronous. As most APIs in SMING framework are not synchronous - but asynchronous utilizing a call -> callback (delegate) structure

There are two ways for accomplishing what you want
1) You keep the client connection open until the result is ready -> but this might lead to a timeout and not a very good design since it will be blocking the execution of other events

2) Your client issues a comment to trigger a scan and will poll every x seconds if the scan is finished

For example the call`s might be:  
`POST /endpoint`  to trigger a refresh - 
`GET /endpoint`  for polling and if the scan finished presenting the results

I am using this technique in a project of mine - I put the code fragment [in a gist ](https://gist.github.com/patrickjahns/74707d1d7361124d5fb0)

You still need to adapt your ajax requests accordingly - but this should get your started with the concept

mxaris, 2016-03-15:
I followed the second way (trigger - polling) and its working fine. thanks

closed 2016-03-15 by mxaris

alonewolfx2, 2016-03-15:
@patrickjahns can you provide PR ?

hreintke, 2016-03-15:
@alonewolfx2 ; 
The proposed solution is a Host based solution -> independent on Sming

patrickjahns, 2016-03-15:
@alonewolfx2 @hreintke 

Is it enough to keep this issue as reference for people wanting the same "feature" or do we need to document it (maybe with a README in the example app?)

hreintke, 2016-03-15:
@patrickjahns : @alonewolfx2 : 
I think it is best to have a wiki page "Limitations" in which features like this are documented.

BTW : Have been thinking if we can keep the httprequest open, and allow the application to send the response at a later time. But that needs thorough investigation.

patrickjahns, 2016-03-15:
@hreintke 

Well you can always delay the closing of the connection until the Clients TCP timeout is reached. But since this is nothing we can control (clients might close it before any result is available) imho think that this is not worth the effort currently. 

Most webdevelopment frameworks already utilize asynchronous structures and I think that people are either used or getting used to this. If someone wishes to have a "real time" communication it would be better served to use tcp/websockets

hreintke, 2016-03-15:
@patrickjahns : 
Agree on the use of tcp/websockets. That is also one of the reasons why I did not actively work on this.