SmingHub/Sming #656: Ajax

teicors, 2016-03-17:
Hi,
I found an issue (may be). I compiled the HttpServer_AJAX from a "out of the box" version of Sming, but I get this error:

[....]
path=/ajax/input
Host === 192.168.1.220
parsed
Request: GET, nodata
TCP received: 436 bytes
onReadyToSendData: 1
response sendHeader
response sendBody
realloc 1 -> 131
realloc 130 -> 260
realloc 259 -> 325
realloc 324 -> 390
realloc 389 -> 455
Stream completed
TCP connection closing
~TCP connection
~TCP connection
3fff2440 already freed
-TCP connection

CONNECTION DROPPED
	(6472)

I reset the esp

dhcp client start...
ip:192.168.1.220,mask:255.255.255.0,gw:192.168.1.120
I'm CONNECTED
addUser: me 123
'/' registered
'/ajax/input' registered
'/ajax/frequency' registered

=== WEB SERVER STARTED ===
192.168.1.220
==============================

pm open,type:2 0

and my browser point to 192.168.1.220 and after few seconds I get a [RST,ACK].

teicors, 2016-03-18:
I updated to the latest version and I'm facing the same issue... After few seconds and button pressed few time, I achieve the same problem :cry: 
And in the same source the ftp server hangs after a 'dir' or 'ls'.

alonewolfx2, 2016-03-18:
@teicors what is your os version and ftp client version ? i had same issue about ftp server.

teicors, 2016-03-19:
@alonewolfx2 here's my system:
kubuntu 15.10
kernel Linux U36SG 4.2.0-34-generic #39-Ubuntu SMP Thu Mar 10 22:13:01 UTC 2016 x86_64 x86_64 x86_64 GNU/Linux
ftp client 0.17-33

the upload and download works perfectly.

hreintke, 2016-04-22:
I am working on an improved version of FTP. 
However that will only be available in the SmingRTOS version.

alonewolfx2, 2016-04-22:
@hreintke i am waiting for yout ftp version ;)

teicors, 2016-04-22:
@hreintke me too !!

hreintke, 2016-05-06:
Please check the flashstream_rboot_ftp branch from SmingRTOS
The ftp server has been updated in that

hreintke, 2016-05-23:
Solved in SmingRTOS

closed 2016-05-23 by hreintke