SmingHub/Sming #658: MQTT client cannot publish message size > 250 bytes

helpme1, 2016-03-17:
I am using SMING v2.1 and node.js Mosca MQTT broker.

I noticed that when publishing a MQTT message > 250 bytes, it will fail and the MQTT connection will sometimes drop. What puzzles me is that the MQTT_MAX_BUFFER_SIZE is set at 1024. So, shouldn't the MQTT client be able to publish a message >250 bytes? How can I modify the code to make it publish message >250 bytes? I am actually targeting a message size of about 1600 bytes.

https://github.com/SmingHub/SmingRTOS/blob/master/sming/sming/network/MqttClient.h

flexiti, 2016-03-17:
It was  fixed  in develop version

helpme1, 2016-03-17:
> It was fixed in develop version
Hi flexiti,

Does it mean that if I modify MQTT_MAX_BUFFER_SIZE to 1600, I can send a mqtt message up to 1600 bytes? Does MQTT_MAX_BUFFER_SIZE include both topic and message size? Or just message?

flexiti, 2016-03-17:
See #579

flexiti, 2016-03-17:
First try if message > 256 and <1024 working for You , next try increase max message size

helpme1, 2016-03-17:
Ok. Thank you very much for your quick response.

closed 2016-03-17 by helpme1