SmingHub/Sming #664: Inconsistencies with AP and STA and EEPROM saving

patrickjahns, 2016-03-23:
Both WifiStation and WifiAccessPoint allow for saving the client data into eeprom or just running currently with the configuration. But WifiStation does not respect the mode of WifiAccessPoint or viceversa.

Either both need to run with `wifi_set_opmode(mode);` or `wifi_set_opmode_current(mode);` 

I propose to have a generalized WifiInterface from which AP and STA will be derived so they respect each other settings.

avr39-ripe, 2016-03-23:
What do you mean "not respect?" There is second argument to WifiStation/WifiAccessPoint.enable which if TRUE tells to SAVE mode to flash.. if no second arg, or second arg FALSE it only sets current opmode..

patrickjahns, 2016-03-23:
I describe the scenario more clearly.

Initially both AP+STA are disabled, now I call `WifiStation.enable(true, false)` the curent running mode would be AP(disabled)+STA(enabled) - if I call now `WifiAccessPoint(true, true)` - it would save both AP+STA as enabled. 

Since this is closely tied together - one should either save it for both or for none - otherwise this behaviour is inconsistent and confusing

avr39-ripe, 2016-03-23:
Yes, all works as you said. Better it will save things separately, but  think it as feature, not bug and resolve it on higher level... why you need AP? for initial setup, while there is no STA connection, right? so as soon as STA is UP and running just disable SOFTAP mode and save to flash STA as current and permanent opmode.. if no STA connection - both STA (trying to connect to preconfigured SSID) is UP and SOFTAP (in case it needs user interaction) is UP.. then using  WifiEvents onSTAGotIP disable SOFTAP :)

hreintke, 2016-03-23:
@avr39-ripe : 
I agree with @patrickjahns. We need to make the API clear and consistent -> should be solved here and not in a "higher level".

As I mentioned also in gitter last night, every Basic_... example should provide easy and understandable code and functionality for newcomers and people who do not know the details of some of the numerous features of Sming. Examples need to be based on this principle, not on the way you me are using Sming. Keep looking "thru the eyes of the user"

avr39-ripe, 2016-03-30:
See #674 for proposed solution

hreintke, 2016-05-23:
Awaiting PR.

closed 2016-05-23 by hreintke