SmingHub/Sming #670: make: *** [out/build/system/m_printf.o] Error 1

Pfannex, 2016-03-25:
Moin,

after installation by using Chocolatey, I try to build sming library.

```
C:\tools\sming\Sming>make all
AS system/irq_check.s
CC system/flashmem.c
CC Services/FATFS/ff.c
CC Services/SpifFS/spiffs_cache.c
CC Services/SpifFS/spiffs_check.c
CC Services/SpifFS/spiffs_gc.c
CC Services/SpifFS/spiffs_hydrogen.c
CC Services/SpifFS/spiffs_nucleus.c
CC Services/SpifFS/spiffs_sming.c
CC Services/libemqtt/libemqtt.c
CC Libraries/Adafruit_GFX/glcdfont.c
CC Libraries/Adafruit_ILI9341/hspi.c
C+ system/esp_cplusplus.cpp
C+ system/m_printf.cpp
In file included from c:\tools\sming\sming\smingcore\network\ntpclient.h:8:0,
                 from c:\tools\sming\sming\smingcore\SystemClock.h:7,
                 from c:\tools\sming\sming\smingcore\smingcore.h:18,
                 from system/m_printf.cpp:10:
c:\tools\sming\sming\smingcore\platform\station.h:36:25: error: 'SC_TYPE_ESPTOUCH_AIRKISS' was not declared in this scope
  SCT_EspTouch_AirKiss = SC_TYPE_ESPTOUCH_AIRKISS
                         ^
make: *** [out/build/system/m_printf.o] Error 1
```

where I´m wrong?

Pfannex, 2016-03-25:
OK, found it....

[https://github.com/SmingHub/Sming/issues/558]

hreintke, 2016-04-21:
Fixed

closed 2016-04-21 by hreintke