SmingHub/Sming #671: Get list of mqtt topics subscribed successfully

helpme1, 2016-03-26:
I am using ESP8266 SMING framework as a MQTT client and using node.js mosca MQTT broker.

I noticed that sometimes, topics are not subscribed successfully. How can I get the list of topics that have been successfully subscribed to the mosca mqtt broker? This way, I can force a re-subscription attempt if the previous attempt failed.

This problem can be easily duplicated with MqttClient_Hello example. Topics will be subscribed successfully the very first time ESP8266 boots up. Now, reset ESP8266 and the topics will be unsubscribed. Now, reset ESP8266 again and the topics will be subscribed successfully.

I am not sure if this is a bug with mosca MQTT broker or ESP8266 or just a normal behavior. One way to work around this problem is to get a list of topics subscribed successfully and force a re-subscription attempt if previous attempts failed.

hreintke, 2016-04-29:
@helpme1 : 
In the current implementation it is not possible to see the successful subscriptions.
As I read in the spec's there is a message type SUBACK which is send when subsribing is OK.

You can take a look at mqttclient.cpp around line 255. If you provide a PR for a solution we can discuss from there.

hreintke, 2016-05-23:
Closing for now.
If you want to provide a PR, please submit in SmingRTOS.

closed 2016-05-23 by hreintke