SmingHub/Sming #672: Bug: The header field name is case-insensitive but is not processed as such.

yunjova, 2016-03-29:
There is a bug in the file: "`Sming/Sming/SmingCore/Network/HttpRequest.cpp`".

The header field name is case-insensitive according to:
  https://tools.ietf.org/html/rfc7230
    ...
    _**3.2.  Header Fields**
       Each header field consists of a case-insensitive field name followed
       by a colon (":"), optional leading whitespace, the field value, and
       optional trailing whitespace._
    ...

So in: "`Sming/Sming/SmingCore/Network/HttpRequest.cpp`"
  the function:
    `String HttpRequest::getHeader(String headerName, String defaultValue /* = "" */)`
  must do a case-insensitive compare.

Example: It should accept `Content-Type:` but also `Content-type:` (lowercase t in -type).

patrickjahns, 2016-03-29:
Good catch. The proposed solution is not complete. There are more parts of the code that need to be addressed in order to enable case insensitive parsing and processing

I will provide a patch towards #622 and #631

patrickjahns, 2016-04-04:
Fix provided - please see #681 and test it

PR #681 (2016-04-05): make header-fields case-insensitive
In order to comply with rfc7230 headerfields are now parsed case-insensitive and retrieved case-insensitive

fixes #672
closed 2016-04-05 by hreintke