SmingHub/Sming #677: Simplify windows installation

Matzz, 2016-04-02:
The recommended windows installation is very monolithic and slow. I think that you should consider that this framework is not the first attempt to use esp8266 for most of the users. Some users may already have MinGW and other stuff configured.
For me the biggest overkill was was downloading eclipse. I couldn't disable this step and your repo was really slow (it took me about 1h to download it where downloading frow official eclipse repo takes about 1m).
I think deprecating manual installation is quite huge mistake. Remember that framework, SDK, IDE and compiler are completely separate things. Using framework should not force us to use specified IDE

patrickjahns, 2016-04-03:
@Matzz 
If you`d read the instructions closely at https://github.com/SmingHub/Sming/wiki/Windows-Quickstart#install-sming you already have an option to omit downloading eclipse:

> If for some reason you don't want Eclipse, UDK, mingw and Environment configured just run:

> choco  install sming.core -y -source 'https://www.myget.org/f/kireevco-chocolatey/'

What is your suggestion for simplifying the installation? 
What concrete steps can be taken to make it easier? 

A while ago was the suggestion to also use platform.io - but we are missing someone with enough knowledge on platform.io to implement it currently

Matzz, 2016-04-05:
Ok sorry I miss that point about eclipse. It would be great if there would be opportunity to install all packages separately. It could be some prompt from installer or just list of dependencies on wiki. Currently it is hard to investigate what is going behind the installation of sming (or maybe I'm not enought experienced in choco).
List of dependent packages is useful when you wish to uninstall sming (eg due to some mess during upgrading). Currently I would like to uninstall all sming related packages except java. No it is hard do guess what was installed.

platform.io - Looks nice. I will try to get a look on it.

patrickjahns, 2016-04-07:
I understand what you are referring too - that`s why I install my stuff by hand. 

The note that this method is deprected is more a general note that there will be not support for people trying to do this and fail (at least that is the way I percieve it).

If you are keen to include support for platform.io that would be really great - also if you can make the installation process via choco easier/better that`d be great

Matzz, 2016-04-10:
BTW I found that http://docs.platformio.org/en/latest/platforms/espressif.html

hreintke, 2016-04-13:
@Matzz : 
Platformio is a good platform to distrubute/give access to Sming environments.
As a start we should look in how we can get Sming hosted on platformio and what requirements platformio has to be able to maintain that.

If you can take the lead in that would be great.

hreintke, 2016-05-23:
Duplicate of the platformio enhancement discussion

closed 2016-05-23 by hreintke