SmingHub/Sming #680: Bug: Spiffs - filename longer than 32 characters.

yunjova, 2016-04-04:
The spiffs build accepts files with names longer than 32 characters (without error/warning).
Accessing these files via the HTTP server gives a `404 - Not found`.

I took me quite a while to find out that the 32-bit limit was the reason why files were not found.

Can you please make an update that deals with this? For instance support longer names in the server or make the build process check on longer names and emit an error or ...

patrickjahns, 2016-04-07:
The limit is set by SPIFFS - specified here:

https://github.com/SmingHub/Sming/blob/49f786e35f5c35305b97d43165cdd97ec8c36cc5/Sming/Services/SpifFS/spiffs_config.h

IMHO changing the value is not needed - 32chars are a lot for a small filesystem like the ESP has.

Related to the filesize - it is actually 31chars, see here: pellepl/spiffs#79

I think this should find a space in FAQ / Documentation so other people don`t have to search as long for this as you did

hreintke, 2016-04-29:
@yunjova,
As @patrickjahns mentiond before, there is no possibility to use longer filenames in spiffs.

I expect you mean the building of the initial FS by spiffy.
If so, can you create a PR for this test,

hreintke, 2016-05-23:
Solved by upgrading to next spiffs in SmingRTOS.
Will not be backported.

closed 2016-05-23 by hreintke