SmingHub/Sming #683: Eclipse resolving standard objects e.g. String

Pfannex, 2016-04-04:
Hi,

I installed all recommended Eclipse/Sming parts inclusive the Espressif SDK.
All functions are running well, also the examples are compilable.

Also I imported the komplete SmingFramework into my Workspace.

Now I got some Errors left over.
E.g. the String object will marked as Error, but is compilable.

Can anybody help me to set up my envirement.

Geetings from Hamburg
Pf@nne

[image]

[image]

cvgaviao, 2016-05-03:
I was getting same problems using eclipse cdt neon at a ubuntu machine.
but after applying the fix in the PR#679 the problem was resolved

hreintke, 2016-05-23:
Issue solved.

closed 2016-05-23 by hreintke