SmingHub/Sming #685: Problems with FTP server

susie-derkin, 2016-04-06:
I have problems on a 512kB ESP8266-07 with the FTP server code. Even the FTPServer_Files example does not really connect to client, everytime when I try to connect via FileZilla I got a reset when the file list should be send:
TCP connected<CR><LF>
send file list: 8<CR><LF>
TCP connection closing<CR><LF>
~TCP connection<CR><LF>
> 226 Transfer Complete.<CR><LF>
<CR><LF>
TCP sent: 395<CR><LF>
<CR><LF>
 ets Jan  8 2013,rst cause:4, boot mode:(3,7)<CR><LF>
<CR><LF>
wdt reset<CR><LF>
load 0x40100000, len 24044, room 16 <CR><LF>
tail 12<CR><LF>
chksum 0x25<CR><LF>
ho 0 tail 12 room 4<CR><LF>
load 0x3ffe8000, len 884, room 12 <CR><LF>
tail 8<CR><LF>
chksum 0x8b<CR><LF>
load 0x3ffe8380, len 5480, room 0 <CR><LF>
tail 8<CR><LF>
chksum 0xde<CR><LF>
csum 0xde<CR><LF>
rlNÿ+TCP connection<CR><LF>
timeout updating: 70 -> 65535<CR><LF>
Server timeout updating: 40 -> 900<CR><LF>

Can anybody confirm my problem or am I doing something wrong?
Any help is welcome.

Thanks 
Alex

teicors, 2016-04-08:
Hi susie-derkin,
I have the same issue. Anyway you can use the server sending and receiving the files, but not listing

Cheers,

susie-derkin, 2016-04-08:
Hi teicors,

thanks for your answer, so I do nothing wrong....
But a problem with the LIST command is not really comfortable, because nearly all graphically based FTP clients will send the LIST command after a successful connect.

I already tried to disable the output of the size of every file (because there were an issue that resulted in an disabled SIZE command and here also fileGetSize was involved) but this does not help....

Is this only a problem on 512k ESPs or is it a general problem?
I can't test it, I only have ESPs with 512k, so maybe anyone with a larger flash size can try it and post the results.

Thanks in advance
Alex

hreintke, 2016-04-09:
Hi,
I think the FTP behavior is somewhat dependent on the client used.
Did some testing with Filezilla and see that the initial connection and LIST is working OK.
But after other activities the LIST is not working properly and also behavior in unclear.

I will be working on other functionalities of FTP shortly and take the above in account.

hreintke, 2016-05-06:
@alonewolfx2 @susie-derkin @teicors

Please check with the FlashStream_rBoot_FTP branch from SminRTOS.
The FTP server has been updated in that

teicors, 2016-05-06:
I'll do some tests this afternoon :)

susie-derkin, 2016-05-22:
I tested the SMING_RTOS example CommandProcessing_Debug with the SMING_RTOS version from today (I saw you merged the branch 'FlashStream_rBoot_FTP'), and...

it works!!!

Many many thanks!!

PS:
I just opened a new issue in SMING_RTOS regarding the SMING_HOME environment variable.

hreintke, 2016-05-23:
Issue resolved.

closed 2016-05-23 by hreintke