SmingHub/Sming #687: SPIFFs/Filesystem issues/remarks

patrickjahns, 2016-04-07:
I`ve noticed a couple of problems / annoyances / issues with the filesystem these days:

- if the file is bigger than free space, we can not recover properly (also see: pellepl/spiffs#82)
- there is currently no simple call in Filesystem to check/get the free space  
  this might be possible by creating a wrapper around `SPIFFS_info` (see: https://github.com/pellepl/spiffs/wiki/Using-spiffs#checking-filesystem)
- methods writing data to spiffs (HttpClient, FTPServer) will simply fail/cause resets when the file system is full or we try to write more data than is available

raburton, 2016-05-11:
Try the newspiffs branch. Don't have the time to test it out on a real device, but integrated the new version and compile tested it at least.

hreintke, 2016-05-23:
New Spiffs version will be integrated within SmingRTOS

closed 2016-05-23 by hreintke