SmingHub/Sming #691: OTA failes on Rocket web server

icodk, 2016-04-10:
OTA with  Rocket 1.2.6 Python/2.7.9 web server failed to transfer complete rom image.
However works correct with nginx-1.8.1.
More observations while testing with the Rocket server (used as development server in web2py framework):
1. Using the same link in a browser the rom0.bin file is correctly downloaded.
2. The  Content-Length header reported by the server is correct.
3. Number of bytes downloaded as reported by rboot is 65536 short of the actual size
4.``` Content-Type === application/octet-stream``` is also correct.
5. spiff_rom.bin  reportedly size wize downloaded  correct. Can't switch to slot 1 because of the error with rom0.bin, therefore  can't verify  that it also function.

From 1 above  ota don't like the Rocket web server for some reason. Browser don't mind.
Thanks for help

raburton, 2016-04-11:
`3. Number of bytes downloaded as reported by rboot is 65536 short of the actual size`

Just to clarify point 3 - rBoot does not do the downloading, or reporting of download sizes, this is higher level code in Sming which goes on to pass the downloaded data to the rBoot api. I suspect this is a bug in the HttpClient class (or in the python server causing a faulty http stream).

patrickjahns, 2016-04-11:
It would be interesting to capture the traffic in both scenarios and compare the differences. (In header fields, data transmitted etc.)

icodk, 2016-04-11:
@raburton
Sorry you are right. Don't blame the messenger:-)

patrickjahns, 2016-04-11:
@icodk

can you capture the traffic and provide it to see whats going on ?

icodk, 2016-04-11:
Will try tomorrow

hreintke, 2016-04-22:
@icodk : 
Any news on this ?

icodk, 2016-04-23:
Not really
Works fine with Apache
I still plan to test the Rocket issue. Haven't had the time yet.

patrickjahns, 2016-04-29:
I have experienced something similar with using Python + Flask and rBootHttpUpdate

```
Download file:
    (0) http://192.168.2.115/rom0.bin -> 2000
Download: http://192.168.2.115:80/rom0.bin
connect to: 192.168.2.115
TcpConnection::connect result:, 0
realloc 45 -> 176
OnConnected
TCP connected
onReadyToSendData: 0
TcpClient request completed
TCP sent: 47
onReadyToSendData: 2
TCP received: 17 bytes
onReadyToSendData: 1
Header pos: 351
Content-Type === application/octet-stream
Last-Modified === Thu, 28 Apr 2016 22:03:48 GMT
Cache-Control === no-cache, public, max-age=43200
Expires === Fri, 29 Apr 2016 10:05:22 GMT
ETag === "flask-1461881028.31-339440-1188567357"
Date === Thu, 28 Apr 2016 22:05:22 GMT
Access-Control-Allow-Origin === *
Server === Werkzeug/0.11.4 Python/2.7.11
TCP received: 1460 bytes
onReadyToSendData: 1

 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x40100000, len 1948, room 16 
tail 12
chksum 0x37
ho 0 tail 12 room 4
load 0x3ffe8000, len 772, room 12 
tail 8
chksum 0x6c
csum 0x6c
```

Randomly the connection resets after the first couple of packets - when I find the time I will investigate this further

@icodk 
it would be also great to get some more input from your side

icodk, 2016-04-29:
@patrickjahns 
Simply  haven't had the  time to investigate farther. So far I can get by using nginx.
Will share any information when available. Your error is somewhat different. You get a reset, while mine was an incomplete image,  not qualified or booting from. Boot from old image.

hreintke, 2016-05-23:
No activity, probably no Sming issue.

closed 2016-05-23 by hreintke